stack-of-tasks/pinocchio #2026: Segfault When Getting Frame Placements

alberthli, 2023-08-03:
## Bug description

When getting frame placements on a simple example script, I get flaky errors of different type. I'm wondering whether I'm doing something obviously wrong - at the very least, a descriptive error message should be shown if possible.
```
# error 1
Segmentation fault (core dumped)

# error 2
Bus error (core dumped)

# error 3
corrupted double-linked list
Aborted (core dumped)

# error 4
double free or corruption (!prev)
Aborted (core dumped)
```

## Expected behavior

I should be able to use the `getFramePlacements` function without error using the workflow explained in examples and documentation.

## Reproduction steps

I'm using the following script and Pinocchio version 2.6.19 installed from `pip` and Python 3.10.4. In the directory containing the script, I placed the `ur_description` directory from [example-robot-data](https://github.com/Gepetto/example-robot-data/tree/master/robots/ur_description). I modified the paths of the URDF to respect this local directory structure for reproducibility. A zip file with the below script + the relevant files is attached: [issue.zip](https://github.com/stack-of-tasks/pinocchio/files/12253272/issue.zip)

```
from pathlib import Path

import numpy as np
import pinocchio as pin

# loading model
model_path = "ur_description"
urdf_path = "ur_description/urdf/ur5_robot.urdf"

model = pin.buildModelFromUrdf(str(urdf_path))

# computing collisions
q = pin.randomConfiguration(model)
data = model.createData()

# computing jacobians
frame = pin.Frame(
    "target_frame",
    3,  # arbitrary parent joint index
    0,
    pin.SE3(np.eye(3), np.ones(3)),  # dummy placement
    pin.FrameType.OP_FRAME,
)
model.addFrame(frame)
frame_idx = model.getFrameId("target_frame")

# joint jacobian
# computes FK and the joint jacobians: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#aa94eb0eb9e51486f618e49617a724d9f
pin.computeJointJacobians(model, data, q)

# frame jacobian
# assumes FK has been called first: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a48c4aeb9f48c48a5965f44a3e18c82b2
pin.updateFramePlacements(model, data)

# needs computeJointJacobians to be called first: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a5aa19d265b05aaa782bb24ae4d8894f0
J = pin.getFrameJacobian(model, data, frame_idx, pin.ReferenceFrame.LOCAL)
```

### System

Output of `lscpu`:
```
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   39 bits physical, 48 bits virtual
CPU(s):                          8
On-line CPU(s) list:             0-7
Thread(s) per core:              2
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           142
Model name:                      Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
Stepping:                        10
CPU MHz:                         815.130
CPU max MHz:                     4000.0000
CPU min MHz:                     400.0000
BogoMIPS:                        3999.93
Virtualization:                  VT-x
L1d cache:                       128 KiB
L1i cache:                       128 KiB
L2 cache:                        1 MiB
L3 cache:                        8 MiB
NUMA node0 CPU(s):               0-7
Vulnerability Itlb multihit:     KVM: Mitigation: VMX disabled
Vulnerability L1tf:              Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
Vulnerability Mds:               Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Meltdown:          Mitigation; PTI
Vulnerability Mmio stale data:   Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Retbleed:          Mitigation; IBRS
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; IBRS, IBPB conditional, STIBP conditional, RSB filling, PBRSB-eIBRS Not affected
Vulnerability Srbds:             Mitigation; Microcode
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon
                                  pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcn
                                 t tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgs
                                 base tsc_adjust sgx bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window 
                                 hwp_epp md_clear flush_l1d arch_capabilities
```

jcarpent, 2023-08-03:
Thanks @alberthli, for this very detailed issue with a reproducible example.
I've tried on my Mac, and I could not reproduce your issue.

First step, could you start in a clean Python environment (e.g., you can try with a clean conda env).
Second step, you can share a docker image to help us investigate the bug.

cmastalli, 2023-08-03:
You're creating the Pinocchio data and then modifying the Pinocchio model.
Have you tried to create the Pinocchio data after modifying its model?

jcarpent, 2023-08-03:
Carlos is right, I miss this point. Thanks Carlos.

alberthli, 2023-08-03:
You're completely correct - rookie mistake on my part, thank you! I had decided to add all of these frames in after I had already coded up many other things and forgot to do it before I initialized the data.

closed 2023-08-03 by alberthli