stack-of-tasks/pinocchio #2027: Inconsistencies between crba() and rena()

Mr-Red331, 2023-08-04:
Hi,
This is a question about the inconsistencies between crba() and rena(). I used two ways (rnea and crba) to calculate `tau`, but the result was inconsistent:
1. tau = rnea(model, data, q, v, a, extf)
2. tau = M*a + C*v + G - JT*extf

Here is the specific process:
1. define a set a const random input data: `q, v, a, extf`;
2. calculate matrix M with `crba(model, data, q)`;
3. calculate matrix C with `computeCoriolisMatrix(model, data, q, v,)`;
4. calculate matrix G with `computeGeneralizedGravity(model, data, q)`;
5. calculate matrix J with `computeJointJacobian(model, data, q, joint, J)` and `getJointJacobian(model, data, joint, LOCAL, J)`;
6. calculate `tau_crba =  M*a + C*v + G - JT*extf`;
7. calculate `tau_rnea = rnea(model, data, q, v, a, extf)`;

But, the final result `tau_crba != tau_rnea`.

At the same time, there is an interesting phenomenon: I let `a.setZero()` and the rest `q, v, extf` remains const random. After going through the above steps, I could get the result `tau_crba = tau_rnea`.

So, is there any wrong in calculating `M` or define `a`?

jcarpent, 2023-08-04:
In C++, only the upper part of the mass matrix is filled, for performance issues. You can use the Eigen API to consider the full matrix (M.selfadjointView<Eigen::[Upper](http://eigen.tuxfamily.org/dox/group__enums.html#gga39e3366ff5554d731e7dc8bb642f83cdafca2ccebb604f171656deb53e8c083c1)>())

Mr-Red331, 2023-08-04:
Thanks! This method worked and my problem was solved.

And futhermore, I would be very grateful if you could tell me why this method could solve this problem.

jcarpent, 2023-08-04:
Just display the mass matrix, and you will see that the lower part is full of 0s.

closed 2023-08-04 by jcarpent

Mr-Red331, 2023-08-04:
That means, in C++, the matrix M calculated with crba() is unfilled and incorrect,  and only the filled Matrix M is the correct for calculate dynamics?

fabinsch, 2023-08-04:
This behavior is expected and described [here](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#ad3da115c312299adc15978286aa82b10) in the official pinocchio documentation.

Mr-Red331, 2023-08-04:
Thanks very much, that is very helpful to me!