stack-of-tasks/pinocchio #2028: Installation error with pip on macOS (ARM)

stephane-caron, 2023-08-07:
## Bug description

The following error happened on macOS with an ARM CPU, after a `pip install pin` in a fresh Conda environment:

```
Traceback (most recent call last):
  [...]
  File "/Users/local_user/miniforge3_arm/envs/upkie/lib/python3.8/site-packages/upkie/envs/upkie_servos_env.py", line 21, in <module>
    import pinocchio as pin
  File "/Users/local_user/miniforge3_arm/envs/upkie/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: dlopen(/Users/local_user/miniforge3_arm/envs/upkie/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/pinocchio_pywrap.cpython-38-darwin.so, 0x0002): tried: '/Users/local_user/miniforge3_arm/envs/upkie/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/pinocchio_pywrap.cpython-38-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/local_user/miniforge3_arm/envs/upkie/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/pinocchio_pywrap.cpython-38-darwin.so' (no such file), '/Users/local_user/miniforge3_arm/envs/upkie/lib/python3.8/site-packages/cmeel.prefix/lib/python3.8/site-packages/pinocchio/pinocchio_pywrap.cpython-38-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))
```

## Reproduction steps

Steps to reproduce the behavior:

1. Create a new conda environment: ``conda create -n upkie``
2. Activate it: ``conda activate upkie``
3. Run `pip install pin`
4. In a Python interpreter, run ``import pinocchio as pin``
5. See error

## Additional context

Despite the environment being named "upkie", the error above is not related to [upkie](https://github.com/tasts-robots/upkie) itself, except the fact that one of its environments imports Pinocchio.

### System

- OS: macOS
- CPU: Apple silicon
- Pinocchio version: 2.6.19

nim65s, 2023-08-07:
Thanks for the report. Could you provide your current pip version and the logs of the `pip install pin` command ? I'd like to see the exact name of the downloaded files.

stephane-caron, 2023-08-07:
Unfortunately I got this report as-is and don't have access to the machine where it occurred :see_no_evil: Sharing it here as maybe there is something we should check, but let's close this lead if we can't reproduce or don't find anything.

nim65s, 2023-08-07:
The first reproduce step is not installing a python version. This pip is not from the conda env.
Please use `conda create -n jnrh-2023 python=3.X` where X is 9, 10 or 11.

3.8 is not available on macos arm64.

closed 2023-08-07 by nim65s