stack-of-tasks/pinocchio #2031: ubuntu version

bhtxy0525, 2023-08-11:
@jcarpent  Hi，can I install pinocchio in the ubuntu16.04 with source code form？

jcarpent, 2023-08-11:
You can install it easily with Conda if available.
Why do you still want to use 16.04?

bhtxy0525, 2023-08-11:
Because our team think 16.04 has the best real-time performance with igh EtherCAT master.  So far，i have installed pinocchio with source many times but failed，i can‘t finish the make process，so i should install it through conda？

jcarpent, 2023-08-11:
Yes, if conda works on your OS, it should be the easiest way to install it ;)

bhtxy0525, 2023-08-11:
thank you very much，i will try it. 
In addition, I would like to know if pinocchio is successfully installed through conda, can I directly migrate the installation files to other OS？

jcarpent, 2023-08-11:
What do you mean by `migrate the installation files to other OS`?

bhtxy0525, 2023-08-11:
I mean the other 16.04 version

bhtxy0525, 2023-08-11:
Because other computers （install with 16.04 OS）can't be connected to the Internet. so i want to migrate the installation files

closed 2023-08-11 by jcarpent

nim65s, 2023-08-11:
> Because our team think 16.04 has the best real-time performance with igh EtherCAT master

It would be interesting to share references benchmarks on this topic. 
Without those, this issue sound foolish.

olivier-stasse, 2023-08-11:
If you really want to have your bug fixed upstream I would recommend to have a look to:
https://icube-robotics.github.io/ethercat_driver_ros2/

This is ros2_control ethercat driver built on top of igh. Several industrial robots are already using it. You will have upstream  fixes on ros_rolling. You will also benefit from ros2_control devs. The community is really investing a lot to have things fixed. And AFAIK most of the ros1 control functionalities are covered.

Building something new on top of 16.04 is becoming too difficult to maintain.