stack-of-tasks/pinocchio #2034: install error

bhtxy0525, 2023-08-12:
@jcarpent Hi, sorry to distrub you.
I have followed the linux installation procedure step by step to install Pinocchio
but when I got to step 4 and input:
`sudo apt update`
the terminal feedback the error:
`The repository 'http://robotpkg.openrobots.org/packages/debian/pub xenial Release' does not have a Release file.
Updating from such a repository can't be done securely, and is therefore disabled by default.`

How can I address this issue?

jcarpent, 2023-08-13:
As we mentioned in other issues to you, we have dropped the support of xenial as it has reached its EOL.

closed 2023-08-13 by jcarpent

bhtxy0525, 2023-08-13:
Thanks for your reply.
Can you tell me which issue your mentioned can I refer to？ 
I‘m very hurry to install Pinocchio，but the official installation guidance don’t works.