stack-of-tasks/pinocchio #2042: Renaming `freeflyer_joint` to `root_joint` in buildSampleModelHumanoid

cmastalli, 2023-08-25:
Having two naming standards looks like a design bug. This is not an ideal choice and I propose to rename it to `root_joint`. I expect to use the same naming standard to include a root joint.

If you agree with this issue, I could create a PR handling it.

cmastalli, 2023-08-25:
It seems the `freeflyer_joint` name is hard coded in https://github.com/stack-of-tasks/pinocchio/blob/devel/src/parsers/sample-models.hxx#L320.

You can see the change of the name by running the following code:
```python
import pinocchio
model = buildSampleModelHumanoid()
print(model.names[1])
```
which is different to
```python
import example_robot_data
model = example_robot_data.load("talos").model # this happens in all the models with FF root.
print(model.names[1])
```

jcarpent, 2023-08-25:
Thanks for the clarification !
Please open a PR with the fix. Thanks for your collaboration !

closed 2023-08-28 by jcarpent