stack-of-tasks/pinocchio #2044: getting error when building example

jessicaleu24, 2023-08-29:
## Bug description

I'm getting error
```dockerfile
/home/git/pinocchio/examples/collisions.cpp: In function ‘int main(int, char**)’:
/home/git/pinocchio/examples/collisions.cpp:48:3: error: ‘computeCollisions’ was not declared in this scope
   48 |   computeCollisions(model,data,geom_model,geom_data,q);
      |   ^~~~~~~~~~~~~~~~~
/home/git/pinocchio/examples/collisions.cpp:54:21: error: ‘CollisionResult’ in namespace ‘hpp::fcl’ does not name a type
   54 |     const hpp::fcl::CollisionResult & cr = geom_data.collisionResults[k];
      |                     ^~~~~~~~~~~~~~~
/home/git/pinocchio/examples/collisions.cpp:57:19: error: ‘cr’ was not declared in this scope; did you mean ‘cp’?
   57 |     std::cout << (cr.isCollision() ? "yes" : "no") << std::endl;
      |                   ^~
      |                   cp

```
when I tried to build the example ``collisions.cpp`` by adding the following lines to cmakelist
```dockerfile
add_executable(collision_test
	examples/collisions.cpp)

target_link_libraries(collision_test PRIVATE Eigen3::Eigen)
```

## Expected behavior

I was hoping to generate a executable to run the example, is there any suggestion?
How can I install hpp-fcl correctly? which version?

### System

- OS: [Ubuntu 20.04]
- Pinocchio version: [(tag: v2.6.20)]

nim65s, 2023-08-29:
Hi @jessicaleu24,

This example should already be build, if you enable hpp-fcl and examples:

https://github.com/stack-of-tasks/pinocchio/blob/4512214cf08e765a28542c75bf6378e934622a38/examples/CMakeLists.txt#L42

To get hpp-fcl, you can download and install a binary release for 20.04 with (at your convenience) apt, conda, or pip: https://stack-of-tasks.github.io/pinocchio/download.html (just replace `pinocchio` or `pin` by `hpp-fcl`), or get it from source: https://github.com/humanoid-path-planner/hpp-fcl

jessicaleu24, 2023-08-29:
cool, thank you

closed 2023-08-29 by jessicaleu24