stack-of-tasks/pinocchio #2054: Having a problem building pinocchio3-preview with casadi

Qrpucp, 2023-09-15:
## Bug description

Thank you for your great work!

I'm trying to use pinocchio and casadi, and I know from #1356 that I need to compile `pinocchio3-preview` to get pinocchio3 to support casadi.

But when I followed the instructions in [#1807](https://github.com/stack-of-tasks/pinocchio/issues/1807#issuecomment-1346201244) to compile `pinocchio3-preview`, I got the following error.

```shell
/opt/conda/envs/pin3preview_casadi/include/boost/preprocessor/iteration/detail/limits/local_256.hpp:13:9: 
error: 'get_pytype' is not a member of 'boost::python::converter::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0> >'
   13 |         BOOST_PP_LOCAL_MACRO(0)
```

In order to verify whether this problem was related to my personal configuration, I also tried it in docker and encountered the same error. And I think he encountered the [same error](https://github.com/stack-of-tasks/pinocchio/issues/1807#issuecomment-1600436719) as me.

By the way, does pinocchio2.9x provide the functions of pinocchio3? If so, does this mean I don't need to compile it manually?

## Reproduction steps

I executed the following command based on the [docker-anaconda image](https://hub.docker.com/r/continuumio/anaconda3). It is almost identical to the one in [#1807](https://github.com/stack-of-tasks/pinocchio/issues/1807#issuecomment-1346201244), except that `make` is installed.

- `conda create --name pin3preview_casadi python=3.9` 
- `conda activate pin3preview_casadi`
- `conda install cmake make casadi compilers=1.4.2`
- `conda install Pinocchio --only-deps`
- Clone the code and switch to `pinocchio3-preview` branch
- `cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CASADI_SUPPORT=ON -DBUILD_TESTING=OFF`
- `make`

### Docker image

The image has been uploaded to [DockerHub](https://hub.docker.com/repository/docker/qrpucp/build_pin3_casadi). 

After creating the container, you need to run `conda activate pin3preview_casadi` to activate the virtual environment. The code is under `/root`.

Mr-Y-B-L, 2023-10-20:
Hello, have you solved the problem? I'm having the exact same problem as you.

> ## Bug description
>
> Thank you for your great work!
>
> I'm trying to use pinocchio and casadi, and I know from #1356 that I need to compile `pinocchio3-preview` to get pinocchio3 to support casadi. [...]

> ## Bug description
>
> Thank you for your great work!
>
> I'm trying to use pinocchio and casadi, and I know from #1356 that I need to compile `pinocchio3-preview` to get pinocchio3 to support casadi. [...]

Mr-Y-B-L, 2023-10-20:
Hello, have you solved the problem? I'm having the exact same problem as you.

GianniLunardi, 2023-11-22:
Hi all,
I am trying to intarface pinocchio-casadi with acados, to directly integrate robot dynamics in the OCP formulation. But I have the same problem as reported above and I cannot compile pinocchio3-preview.

Mr-Y-B-L, 2023-11-22:
> Hi all, I am trying to intarface pinocchio-casadi with acados, to directly integrate robot dynamics in the OCP formulation. But I have the same problem as reported above and I cannot compile pinocchio3-preview.

Sorry, I haven't found a  solution yet.But the jnrh-2023 project can help bind Pinocchio3 with Python