stack-of-tasks/pinocchio #2062: I could not use the Pinocchio preview version to load a simple SDF file

XiaoWZENG, 2023-09-24:
I could not use the Pinocchio preview version to load a simple SDF file, could you provide a simple demo which you have tried successfully, so that I could debug the issue better.

Now I really don't known the problem is caused by my fault or the unstability of the pinocchio preview version.

I need your help please,  just a simple demo , thanks  lot.

stephane-caron, 2023-09-25:
To maximize your chances of getting feedback from someone in the community, you should prepare a minimum example showcasing your issue, including (1) a traceback of the error you encountered, (2) the simple SDF file your mention and (3) the minimum amount of code needed to trigger the issue.

When opening a [New issue](https://github.com/stack-of-tasks/pinocchio/issues/new/choose) there is a "Bug report" template you can use. This template is here to help you get good feedback.

EtienneAr, 2023-10-03:
Hello @XiaoWZENG,

I took a look at [pinocchio3-preview](https://github.com/stack-of-tasks/pinocchio/tree/pinocchio3-preview) way to parse SDF. There are indeed some instabilities.
However, everything is patched in the full version of pinocchio3 (that is under development and not released yet).

If you give me more details about your problem, I might be able to help you find a work-around for the current version of pinocchio3-preview : what is the error, how to reproduce, ...

Please re-open the issue in that case.

closed 2023-10-03 by EtienneAr