stack-of-tasks/pinocchio #2063: Python Casadi GeometryModel and GeometryData not working

warrenhanwen, 2023-09-26:
Hi developers, first thank you for implementing such a useful library.

I'm on branch `pinocchio3-preview` and I'm trying to build a casadi collision model and collision data from urdf. In the following Python code I tried two ways to init a casadi `GeometryModel` but neither works
```Python
import pinocchio as pin
from pinocchio import casadi as cpin

urdf_path = ...

pin_model = pin.buildModelFromUrdf(urdf_path, pin.JointModelFreeFlyer())
pin_collision_model = pin.buildGeomFromUrdf(pin_model, urdf_path, pin.GeometryType.COLLISION)

cmodel = cpin.Model(pin_model)
# Init Method 1
c_collision_model = cpin.GeometryModel(pin_collision_model)
# Init Method 2
c_collision_model = cpin.buildGeomFromUrdf(cmodel, urdf_path, pin.GeometryType.COLLISION)
```

Init Method 1 gives me this error:
```
Boost.Python.ArgumentError: Python argument types in
    pinocchio.pinocchio_pywrap_default.buildGeomFromUrdf(Model, str, GeometryType)
did not match C++ signature:
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, pinocchio::GeometryModel {lvalue} geom_model, boost::shared_ptr<hpp::fcl::MeshLoader> mesh_loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, boost::shared_ptr<hpp::fcl::MeshLoader> mesh_loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, pinocchio::GeometryModel {lvalue} geom_model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > package_dir, boost::shared_ptr<hpp::fcl::MeshLoader> mesh_loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > package_dir, boost::shared_ptr<hpp::fcl::MeshLoader> mesh_loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, pinocchio::GeometryModel {lvalue} geom_model, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > package_dirs, boost::shared_ptr<hpp::fcl::MeshLoader> mesh_loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > package_dirs, boost::shared_ptr<hpp::fcl::MeshLoader> mesh_loader)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, pinocchio::GeometryModel {lvalue} geom_model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > package_dir)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > package_dir)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, pinocchio::GeometryModel {lvalue} geom_model)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, pinocchio::GeometryModel {lvalue} geom_model, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > package_dirs)
    buildGeomFromUrdf(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > urdf_filename, pinocchio::GeometryType geom_type, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > package_dirs)
```

Init Method 2 gives me this error:
```
Boost.Python.ArgumentError: Python argument types in
    GeometryModel.__init__(GeometryModel, GeometryModel)
did not match C++ signature:
    __init__(_object* self)
```

I'm pretty sure I installed pinocchio with casadi support because when I use my IDE's `Ctrl+Left Click` to check the source code of `cpin.Model`, it is indeed from `pinocchio_pywrap_casadi`.

But when I used the same trick to check the source of `cpin.GeometryModel`, `cpin.GeometryData`, and `cpin.buildGeomFromUrdf`, I found that the first two come from `pinocchio_pywrap_default` not `pinocchio_pywrap_casadi`. The third one comes from `~/miniconda3/envs/my_env/lib/python3.9/site-packages/pinocchio/deprecated.py`.

Does this indicate that my pinocchio version doesn't contain Casadi support for `GeometryModel` and `GeometryData`? If I want to use them for collision avoidance in optimization-based control, what should I do?

Thank you!

jcarpent, 2023-09-26:
Dear @warrenhanwen,

First of all, thanks for your interest in Pinocchio and its new features.
Collision detection involves iterative algorithms for finding the closest contact points and the normal (see our [recent paper](https://hal.science/hal-04070039/document) on the topic for a complete review). In other words, automatic differentiation is likely to fail or provide a wrong gradient.
This is why we do not provide automatic differentiation support for collision geometries, which will also imply updating hpp-fcl accordingly to be fully templatized by the scalar type.
In the meanwhile, we have recently published a [paper](https://hal.science/hal-03780482/document) for efficiently and precisely evaluating collision derivatives. It will be fully integrated in Pinocchio soon.

Best,
Justin

closed 2023-09-26 by jcarpent

fabinsch, 2023-09-26:
Hi @warrenhanwen,
you can have a look at this [notebook](https://github.com/nmansard/jnrh2023/blob/main/4_with_obstacles.ipynb) which explains how you can still use Pinocchio and casadi to solve an optimization problem under collision avoidance constraints. It does not work directly with the geometry model, as @jcarpent explained above. But there is an approximation that you can do, expressing the collision object as the encapsulating ellipsoid and work with casadi and Pinocchio on these objects. This notebook, and also the preceding ones, can help you to formulate your problem in the meantime.