stack-of-tasks/pinocchio #2069: REGEX REPLACE in CMakeLists.txt fails for complex directory paths

Pathorse, 2023-10-17:
## Bug description

Pinocchio build fails when using complex directory paths, e.g., `/test_pinnochio_build/1.0+gitAUTOINC+8ajb23j4b/pinocchio/build`, [as  REGEX REPLACE fails to correctly amend strings before attempting to create symbolic links (or copying)](https://github.com/stack-of-tasks/pinocchio/blob/903cdb8b1fd3bfc21779e76b1c97dbefe5e690da/CMakeLists.txt#L269).

This issue was found while using pinocchio with the Yocto SDK.

## Expected behavior

The build should not fail for any directory paths given.

## Reproduction steps

Steps to reproduce the behavior:
1. Create a test folder:
```bash
mkdir -p test_pinnochio_build/1.0+gitAUTOINC+8ajb23j4b
```
2. Clone Pinocchio
```bash
git clone git@github.com:stack-of-tasks/pinocchio.git
```
3. Initialize submodules
```bash
cd  test_pinnochio_build/1.0+gitAUTOINC+8ajb23j4b/pinocchio
git submodule update --init
```
4. Build
```bash
mkdir build && cd build
cmake ..
```
5. See error

## Solution

The problem can be solved by replacing `REGEX REPLACE` with `REPLACE`, which given the fact that the desired string operation doesnt require the flexibility of regular expressions may make sense to move away from anyways (unless I am missing something here which might be the case as I am pretty new with Pinocchio). The swap must be done in three files: 

[1. Toplevel CMakeLists.txt](https://github.com/stack-of-tasks/pinocchio/blob/4512214cf08e765a28542c75bf6378e934622a38/CMakeLists.txt#L269)
[2. Python bindings CMakeLists.txt](https://github.com/stack-of-tasks/pinocchio/blob/4512214cf08e765a28542c75bf6378e934622a38/bindings/python/CMakeLists.txt#L26)
[3. Unittest CMakelists.txt](https://github.com/stack-of-tasks/pinocchio/blob/4512214cf08e765a28542c75bf6378e934622a38/unittest/CMakeLists.txt#L30)

### System

- OS: Ubuntu 22.04
- Pinocchio version: 2.6.20

jcarpent, 2023-10-17:
Thanks @Pathorse for the nice report. @nim65s @jorisv Could you have a look to this issue in more depth?

nim65s, 2023-10-17:
I think we should simply switch to a more standard CMake use, and:
1. stop simlinking stuff around
2. list all header / sources in CMakeLists.txt

If agreed, I probably can work on it this week, but probably it would be better to do this on the private branch for v3

jorisv, 2023-10-25:
@Pathorse This issue should be fixed by #2070

manumerous, 2023-10-25:
Thanks a lot!

jcarpent, 2023-10-26:
Solve via #2070