stack-of-tasks/pinocchio #2071: Python bindings changing type depending on imports

joao-pm-santos96, 2023-10-19:
Hello all, I've found what I think is an issue that I would like your comments, please.

## Bug description

In a python script, I am trying to import both pinocchio and ompl, which I am able to. The strange behavior that I am witnessing is that, depending on which one I import first, the types of the variables within pinocchio AND ompl change! 

## Expected behavior

I was expecting them to don't change.

## Reproduction steps

When I run 

```py
from os.path import abspath, dirname, join
from sys import argv

from ompl import base, geometric
import pinocchio

# This path refers to Pinocchio source code but you can define your own directory here.
pinocchio_model_dir = join(dirname(dirname(str(abspath(__file__)))), "models")

# You should change here to set up your own URDF file or just pass it as an argument of this example.
urdf_filename = (
    pinocchio_model_dir
    + "/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf"
    if len(argv) < 2
    else argv[1]
)

# Load the urdf model
model = pinocchio.buildModelFromUrdf(urdf_filename)
# print("model name: " + model.name)

# Create data required by the algorithms
data = model.createData()

print("Pinocchio model.names type:", end=" ")
print(type(model.names))
print("")

bounds = base.RealVectorBounds(model.nq)
print("OMPL base.RealVectorBounds.high type:", end=" ")
print(type(bounds.high))
print("")
```

I expected the output to be 

```
Pinocchio model.names type: <class 'pinocchio.pinocchio_pywrap.StdVec_StdString'>

OMPL base.RealVectorBounds.high type: <class 'ompl.util._util.vectorDouble'>
```

Yet the actual output is 

```
Pinocchio model.names type: <class 'ompl.util._util.vectorString'>

OMPL base.RealVectorBounds.high type: <class 'ompl.util._util.vectorDouble'
```

and if switch from 

```py
from ompl import base, geometric
import pinocchio
```

to

```py
import pinocchio
from ompl import base, geometric
```

the output becomes

```
Pinocchio model.names type: <class 'pinocchio.pinocchio_pywrap.StdVec_StdString'>

OMPL base.RealVectorBounds.high type: <class 'pinocchio.pinocchio_pywrap.StdVec_Double'
```

## Additional context

### System

- OS: ubuntu 22.04
- Pinocchio version: 2.6.20

nim65s, 2023-10-20:
Hi @joao-pm-santos96,

Do you have any issue other than the name changes ?

I guess ompl is also using boost python to define how to translate some simple stuff between C++ and Python, and it might be defined in the same way and do the same thing, so changing the name wouldn't be an issue.

jcarpent, 2023-10-20:
As mentioned by @nim65s, behind it will be the same C++ objects. This is indeed related to Boost.Python: it just registers only type to avoid duplicate symbols and incompatibility issues.

joao-pm-santos96, 2023-10-20:
Hi @nim65s and @jcarpent . Once again your speed of response was incredible. Thank you!

> Do you have any issue other than the name changes ?

I've found this behavior because if I run 

```py
jointsToLockIDs = [1, 2, 3]
initialJointConfig = np.zeros(model.nq)
model_reduced = pinocchio.buildReducedModel(model, jointsToLockIDs, initialJointConfig)
``` 

when I first import ompl and only after pinocchio, I get the error

```
Exception has occurred: ArgumentError
Python argument types in
    pinocchio.pinocchio_pywrap.buildReducedModel(Model, list, numpy.ndarray)
did not match C++ signature:
    buildReducedModel(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::vector<pinocchio::GeometryModel, Eigen::aligned_allocator<pinocchio::GeometryModel> > list_of_geom_models, std::vector<unsigned long, std::allocator<unsigned long> > list_of_joints_to_lock, Eigen::MatrixBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> > reference_configuration)
    buildReducedModel(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::GeometryModel geom_model, std::vector<unsigned long, std::allocator<unsigned long> > list_of_joints_to_lock, Eigen::MatrixBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> > reference_configuration)
    buildReducedModel(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::vector<unsigned long, std::allocator<unsigned long> > list_of_joints_to_lock, Eigen::MatrixBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> > reference_configuration)
  File "/home/joao/test/main.py", line 39, in <module>
    model_reduced = pinocchio.buildReducedModel(model, jointsToLockIDs, initialJointConfig)
Boost.Python.ArgumentError: Python argument types in
    pinocchio.pinocchio_pywrap.buildReducedModel(Model, list, numpy.ndarray)
did not match C++ signature:
    buildReducedModel(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::vector<pinocchio::GeometryModel, Eigen::aligned_allocator<pinocchio::GeometryModel> > list_of_geom_models, std::vector<unsigned long, std::allocator<unsigned long> > list_of_joints_to_lock, Eigen::MatrixBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> > reference_configuration)
    buildReducedModel(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::GeometryModel geom_model, std::vector<unsigned long, std::allocator<unsigned long> > list_of_joints_to_lock, Eigen::MatrixBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> > reference_configuration)
    buildReducedModel(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, std::vector<unsigned long, std::allocator<unsigned long> > list_of_joints_to_lock, Eigen::MatrixBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> > reference_configuration)
```

which does not happen if I import them the other way around. I just don't know which problems I will potentially encounter further down the line...

joao-pm-santos96, 2023-10-20:
> As mentioned by @nim65s, behind it will be the same C++ objects. This is indeed related to Boost.Python: it just registers only type to avoid duplicate symbols and incompatibility issues.

One thing that differs is that most of your implementations of `std::vector` have the `.tolist()` method while ompl's don't. This obviously is a non issue since we cant use python's `list()` but is just one more difference that I don't know if they will or wont create problems further down the line.

jcarpent, 2023-10-20:
You're right. We can work on adding it when ompl is called first. It will take some time because we have a limited bandwidth.
@jorisv If you are interested, I can explain you how to do this properly in EigenPy

joao-pm-santos96, 2023-10-20:
Awesome, you guys are amazing, thank you!

jorisv, 2023-10-23:
@jcarpent I'm interested in working on it. I will create a MR soon.