stack-of-tasks/pinocchio #2072: pinocchio.XYZQUATToSE3(np.array([ ]) takes in any size of the array

thanhndv212, 2023-10-31:
Hi.
Apparently, as I understand, the function `pinocchio.XYZQUATToSE3(np.array([ ])` should take in only numpy.array which has a shape of (7, ), but strangely it returns regardless of the numpy.array input. 

An error should be given when the shape of the array is incorrect. I suppose it could be fixed in `pinocchio_pywrap`
```
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio as pin
pin.XYZQUATToSE3(np.array([1])))
print(pin.XYZQUATToSE3(np.array([1, 2, 3])))
print(pin.XYZQUATToSE3(np.array([1, 2, 3, 4, 5])))>>> import numpy as np
>>> 
>>> print(pin.XYZQUATToSE3(np.array([1])))
  R =
         -inf  4.74124e-127  1.05712e+181
 4.74124e-127             1  1.72355e-141
-1.05712e+181  1.72355e-141          -inf
  p =            1            0 1.42202e-319

>>> print(pin.XYZQUATToSE3(np.array([1, 2, 3])))
  R =
 -9 -12   6
 12  -7   4
 -6   4  -1
  p = 1 2 3

>>> print(pin.XYZQUATToSE3(np.array([1, 2, 3, 4, 5])))
  R =
-49 -28  46
 52 -39   4
-14  44 -25
  p = 1 2 3

```

jcarpent, 2023-10-31:
@jorisv When you have time, it would be nice to look at this issue.

jcarpent, 2023-11-01:
Solved via #2073

closed 2023-11-01 by jcarpent