stack-of-tasks/pinocchio #2077: Memory leak when using buildGeomFromUrdfString

jmirabel, 2023-11-07:
Line
https://github.com/stack-of-tasks/pinocchio/blob/4512214cf08e765a28542c75bf6378e934622a38/bindings/python/parsers/urdf/geometry.cpp#L174
ties the lifetime of the return value to the lifetime of the optional argument "geometry_model". As a consequence, when the argument is not specified by the caller (python side), the value returned by the function is not tied to any python object and therefore the return value, initially dynamically allocated by buildGeomFromUrdfString, is never freed.

@PierreMinotti FYI

jcarpent, 2023-11-07:
@jmirabel Could you provide a reproducible example?

jmirabel, 2023-11-07:
Here is it:
```python
import pinocchio
import os

urdf_path = "/opt/openrobots/share/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf"
dirs = [ "/opt/openrobots/share" ]

for i in range(10000):
    model = pinocchio.buildModelFromUrdf(urdf_path)
    if False:
        # version with mem leak
        gmodel = pinocchio.buildGeomFromUrdf(
            model,
            urdf_path,
            pinocchio.COLLISION,
            package_dirs=dirs,
        )
    else:
        # version without mem leak
        gmodel = pinocchio.GeometryModel()
        gmodel = pinocchio.buildGeomFromUrdf(
            model,
            urdf_path,
            pinocchio.COLLISION,
            gmodel,
            package_dirs=dirs,
        )
```

jcarpent, 2023-11-07:
@jorisv Do you have time to handle this issue?

jorisv, 2023-11-07:
I will work on it

PR #2082 (2023-11-13): Topic/urdf memory leak
close #2077
closed 2023-11-17 by jorisv