stack-of-tasks/pinocchio #2078: Usage of Casadi SX with Pinocchio

UM-A, 2023-11-07:
Hi, I think this question has been asked previously and there were some answers. I want to go from my urdf file to Casadi SX robot equations. As far as I understand, it might be possible to do with pinocchio. However, when I try to use those expressions I get errors. I am sharing the code snippet, in case I am doing something wrong. Looking forward to any responses.

Code:
model=pinocchio.buildModelFromUrdf("RobotURDF.urdf" )
data=model.createData()
q = SX.sym("q", model.nq)

M = pinocchio.computePotentialEnergy(model, data, np.array([q[0],q[1],q[2],q[3]]))

Error:
Python argument types in
    pinocchio.pinocchio_pywrap.computePotentialEnergy(Model, Data, numpy.ndarray)
did not match C++ signature:
    computePotentialEnergy(struct pinocchio::ModelTpl<double,0,struct pinocchio::JointCollectionDefaultTpl> model, struct pinocchio::DataTpl<double,0,struct pinocchio::JointCollectionDefaultTpl> {lvalue} data)
    computePotentialEnergy(struct pinocchio::ModelTpl<double,0,struct pinocchio::JointCollectionDefaultTpl> model, struct pinocchio::DataTpl<double,0,struct pinocchio::JointCollectionDefaultTpl> {lvalue} data, class Eigen::MatrixBase<class Eigen::Matrix<double,-1,1,0,-1,1> > q)

jorisv, 2023-11-08:
Hello @UM-A,

Pinocchio 2 is only supporting Casadi in C++. The Python support will be added in Pinocchio 3 that is not released and not public.

Hopefully a pre-release is available. You can find instruction in this [tutorial](https://github.com/nmansard/jnrh2023/tree/main).
I recommend you to install it with conda (in a new environment), then to run the [setup notebook](https://github.com/nmansard/jnrh2023/blob/main/0_setup.ipynb) to test your installation.

Others notebook will show you how to use the Pinocchio Casadi binding (`from pinocchio import casadi as cpin`).

aky-u, 2025-05-05:
Hello @jorisv,
I'm using the ROS 2 Humble version of Pinocchio and encountered an error when trying to import pinocchio.casadi.
Does this mean that cpin is not yet supported in this version? 
I would appreciate any clarification you could provide. Thank you! 
(As a second option, I'm using C++ instead, just for your information.)

```bash
 $python -c "from pinocchio import casadi as cpin"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'casadi' from 'pinocchio' (/opt/ros/humble/lib/python3.10/site-packages/pinocchio/__init__.py)
```

nim65s, 2025-05-05:
@aky-u : this is another issue, let us move to #2674

closed 2025-05-05 by jcarpent