stack-of-tasks/pinocchio #2085: CppAd & Codegen test compile failures

ryanelandt, 2023-11-14:
Pinocchio: 2.6.20
Boost: 1.74.0
Eigen: 3.4.0
Ubuntu: 22.04.3

Similar to #1999.

I can compile from source on master. When I modify the CMakeLists.txt file and change `BUILD_WITH_AUTODIFF_SUPPORT` and `BUILD_WITH_CODEGEN_SUPPORT` to `ON` I get the compile error below.

```
/usr/include/boost/math/constants/constants.hpp:265:3: error: no matching function for call to ‘boost::math::constants::detail::constant_pi<CppAD::AD<CppAD::cg::CG<double> > >::get(boost::math::constants::construction_traits<CppAD::AD<CppAD::cg::CG<double> >, boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> >::type)’
  265 |   BOOST_DEFINE_MATH_CONSTANT(pi, 3.141592653589793238462643383279502884e+00, "3.14159265358979323846264338327950288419716939937510582097494459230781640628620899862803482534211706798214808651e+00")
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ryane/dev_model/pinocchio/build/include/pinocchio/codegen/cppadcg.hpp:15,
                 from /home/ryane/dev_model/pinocchio/unittest/cppadcg-joint-configurations.cpp:5:
/home/ryane/dev_model/pinocchio/build/include/pinocchio/autodiff/cppad.hpp:35:34: note: candidate: ‘template<int N> static boost::math::constants::detail::constant_pi<CppAD::AD<Base> >::ADScalar boost::math::constants::detail::constant_pi<CppAD::AD<Base> >::get(const mpl_::int_<N>&) [with int N = N; Scalar = CppAD::cg::CG<double>]’
   35 |           static inline ADScalar get(const mpl::int_<N>& n)
      |                                  ^~~
/home/ryane/dev_model/pinocchio/build/include/pinocchio/autodiff/cppad.hpp:35:34: note:   template argument deduction/substitution failed:
In file included from /home/ryane/dev_model/pinocchio/build/include/pinocchio/math/fwd.hpp:10,
                 from /home/ryane/dev_model/pinocchio/build/include/pinocchio/codegen/cppadcg.hpp:10,
                 from /home/ryane/dev_model/pinocchio/unittest/cppadcg-joint-configurations.cpp:5:
/usr/include/boost/math/constants/constants.hpp:265:3: note:   ‘boost::math::constants::construction_traits<CppAD::AD<CppAD::cg::CG<double> >, boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> >::type’ {aka ‘boost::integral_constant<int, 0>’} is not derived from ‘const mpl_::int_<N>’
  265 |   BOOST_DEFINE_MATH_CONSTANT(pi, 3.141592653589793238462643383279502884e+00, "3.14159265358979323846264338327950288419716939937510582097494459230781640628620899862803482534211706798214808651e+00")
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ryane/dev_model/pinocchio/build/include/pinocchio/codegen/cppadcg.hpp:15,
                 from /home/ryane/dev_model/pinocchio/unittest/cppadcg-joint-configurations.cpp:5:
/home/ryane/dev_model/pinocchio/build/include/pinocchio/autodiff/cppad.hpp:41:34: note: candidate: ‘template<class T, T value> static boost::math::constants::detail::constant_pi<CppAD::AD<Base> >::ADScalar boost::math::constants::detail::constant_pi<CppAD::AD<Base> >::get(const std::integral_constant<T, value>&) [with T = T; T value = value; Scalar = CppAD::cg::CG<double>]’
   41 |           static inline ADScalar get(const std::integral_constant<T, value> &n)
      |                                  ^~~
/home/ryane/dev_model/pinocchio/build/include/pinocchio/autodiff/cppad.hpp:41:34: note:   template argument deduction/substitution failed:
In file included from /home/ryane/dev_model/pinocchio/build/include/pinocchio/math/fwd.hpp:10,
                 from /home/ryane/dev_model/pinocchio/build/include/pinocchio/codegen/cppadcg.hpp:10,
                 from /home/ryane/dev_model/pinocchio/unittest/cppadcg-joint-configurations.cpp:5:
/usr/include/boost/math/constants/constants.hpp:265:3: note:   ‘boost::math::constants::construction_traits<CppAD::AD<CppAD::cg::CG<double> >, boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> >::type’ {aka ‘boost::integral_constant<int, 0>’} is not derived from ‘const std::integral_constant<T, value>’
  265 |   BOOST_DEFINE_MATH_CONSTANT(pi, 3.141592653589793238462643383279502884e+00, "3.14159265358979323846264338327950288419716939937510582097494459230781640628620899862803482534211706798214808651e+00")
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

ryanelandt, 2023-11-15:
The fix in #2000 doesn't appear to be effective because of the renaming in [Boost 1.77](https://github.com/boostorg/math/commit/5ae92ba0c31eab0e1f0134ef144ad51e3a34b455#diff-b00f0c15ac166c7236575aa1e8b013fe357eb2f73159f6d2c199fd2f9522ad6eR21).

I'll test tomorrow to see if the following might work:

```c++
template <class AnyT>
static inline ADScalar get(const AnyT &n)
{
  return ADScalar(constant_pi<Scalar>::get(n));
}
```

ARBITER1994, 2024-01-09:
> The fix in #2000 doesn't appear to be effective because of the renaming in [Boost [...]

Thank you so much.
I can successfully compile and pass the cppad/cppadcg-joint-configurations.cpp. 
Pinocchio: 2.6.21
Boost: 1.74.0
Eigen: 3.3.0
Ubuntu: 22.04.3

PR #2132 (2024-01-16): Fix build with Boost < 1.77
Close #2085

We must define function with std::integral_constant with Boost >= 1.77 and boost::integral_constant with Boost < 1.77
closed 2024-01-24 by jorisv