stack-of-tasks/pinocchio #2091: Compilation problem pinocchio3-preview with casadi support + python bindings

lvanroye, 2023-11-21:
## Bug description

When I try to compile pinocchio3-preview with casadi support + python bindings, I get a compilation error: 

in  pinocchio/bindings/python/algorithm/expose-com.cpp:5:
/usr/local/include/eigenpy/eigen-from-python.hpp:35:8: error: ambiguous template instantiation for ‘struct eigenpy::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>, Eigen::MatrixBase<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1> > >’

I have the most recent version (on git) of eigenpy installed and as well as the newest version of casadi.
I compiled with options: -DBUILD_WITH_CASADI_SUPPORT=ON -DBUILD_PYTHON_INTERFACE=ON.
[compile_output.txt](https://github.com/stack-of-tasks/pinocchio/files/13425046/compile_output.txt)

I included all compilation output in the attachments. I also included the docker file that can used to reproduce the compilation error. 

I'd really appreciate some help in finding my error in configuration or a possible bug.

Best,
Lander

### Docker image

This docker file can be used to reproduce the rror
```dockerfile
FROM ubuntu:22.04

ARG DEBIAN_FRONTEND=noninteractive

RUN apt-get update -y && apt-get upgrade -y

RUN apt install git wget -y python-is-python3
RUN apt install gfortran liblapack-dev pkg-config cmake gcc --install-recommends -y
RUN apt install swig -y
RUN apt install g++ -y
RUN apt install python3-pip -y
RUN python3 -m pip install matplotlib
RUN apt install pybind11-dev -y
RUN git clone https://github.com/casadi/casadi.git && cd casadi && mkdir build
RUN cd casadi/build && cmake \
    -DWITH_IPOPT=OFF -DWITH_BUILD_IPOPT=OFF\
    -DWITH_BUILD_MUMPS=OFF -DWITH_BUILD_METIS=OFF\
    -DWITH_OPENMP=OFF -DWITH_THREAD=OFF -DWITH_PYTHON=ON -DWITH_PYTHON3=ON -DPYTHON_PREFIX=/usr/local/lib/python3.10/dist-packages ..
RUN cd casadi/build && make -j && make install
RUN apt install libboost-all-dev -y
RUN git clone https://github.com/stack-of-tasks/eigenpy.git && cd eigenpy && mkdir build
RUN cd eigenpy/build && cmake -DCMAKE_INSTALL_INCLUDEDIR=include/eigenpy ..
RUN cd eigenpy/build && make -j && make install
RUN apt install libeigen3-dev
RUN apt install liburdfdom-dev -y
RUN git clone https://github.com/stack-of-tasks/pinocchio.git --recursive && cd pinocchio && git checkout pinocchio3-preview && mkdir build
RUN cd pinocchio/build && cmake -DBUILD_WITH_CASADI_SUPPORT=ON -DBUILD_PYTHON_INTERFACE=ON -DCMAKE_CXX_FLAGS=-w ..
RUN cd pinocchio/build && make -j 4 && make install

```

lvanroye, 2024-01-04:
Hello,

I'll use that workaround now then. Thanks a lot it is really helpful!

Lander

jorisv, 2024-01-15:
Thank you @xeonz1,

I am closing this issue because we do not maintain this branch.
The fix will be done in Pinocchio 3.

closed 2024-01-15 by jorisv