stack-of-tasks/pinocchio #2093: Discrepancy between Centroidal Dynamics derivations

Michiel283, 2023-11-22:
First of all, I would like to thank you for developing the Pinocchio library as it has helped me a lot during my research. I am currently trying to verify my Centroidal Dynamics model that I derived analytically, by comparing my result with the result obtained from Pinocchio. I expected the results to be identical, but this doesn't seem to be the case, and I was hoping someone could help me.

I derived an analytical expression for the CMM matrix using the David E. Orin's 2008 paper "_Centroidal Momentum Matrix of a humanoid robot: Structure and properties_". When I check my CMM matrix with the CMM matrix computed by ccrba algorithm, the two matrices are identical. Both matrices have the following block structure:
[image]

However, when I compute dAg, you would expect that the time-derivative of Ag would have a similar block-structure as is also the case in my analytical derivation. When I compute the dAg matrix using dccrba or the computeCentroidalMapTimeVariation function, only the rows belonging to the linear momentum are identical. Moreover, the block-matrix which intuitively would be filled with zero's is nonzero in the dAg matrix computed by dccrba. 

When deriving the centroidal momentum hg, I derived the following analytic expression:
[image]

where A_B is the CMM matrix that maps the generalized joint rates to the momentum in the **body** frame, and X_G transforms this CMM matrix to the Centroidal frame G. Now, when computing the time-derivative of the centroidal momentum dhg, and thus also the time-derivative of the CMM, you would expect the following expression for dAg:
[image]

However, when looking at pinocchio computation, the time-derivative seems to be computed in a different way. From my understanding, it seems like the translational transformation is neglected when taking the time-derivative of Ag, and only the rotational transformation is taken into account. To my understanding, the expression below is later transformed to the COM position, which would explain the non-zero values in the dAg matrix.
[image]

jcarpent, 2023-11-22:
Thanks @Michiel283 for raising this issue. 
I've provided a fix in #2094.

Michiel283, 2023-11-23:
@jcarpent Thank you so much for the quick response and providing the fix!

jcarpent, 2023-11-23:
Solved via #2094

closed 2023-11-23 by jcarpent

Michiel283, 2023-12-08:
Hi @jcarpent ,

Thank you for releasing a new version of Pinocchio with the CMM correction. I tested the CMM matrices for my system and compared them with my analytical derivation and all matrices were identical. I then computed the centroidal momentum and its time-derivative using the computeCentroidalMomentum(TimeVariation), but the $\dot{\mathbf{h}}_\mathrm{G}$ did not match with my analytic derivation. I think it might be caused by the time-derivative of the transformation to the COM frame again. 
```math
\dot{\mathbf{h}}_\mathrm{G} = {}^\mathrm{G}\mathbf{X}^*_\mathrm{I}\dot{\mathbf{h}}_\mathrm{I} +{}^\mathrm{G}\dot{\mathbf{X}}^*_\mathrm{I}\mathbf{h}_\mathrm{I} \neq {}^\mathrm{G}\mathbf{X}^*_\mathrm{I}\dot{\mathbf{h}}_\mathrm{I}
```
However, the linear part of $\dot{\mathbf{h}}_\mathrm{G}$ is independent on the chain-rule correction above, but this also doesn't match with my analytical derivation. Do you have a suggestion which might explain the discrepancy in the linear centroidal momentum time-variation?