stack-of-tasks/pinocchio #2105: Adjusting the model frame

abuibaid, 2023-11-29:
Hi there, 

I have just learned about this wonderful tool. I have gone through and It appears it can do multi things such as kinematics and dynamics which I'm really interested in! But I have faced one problem which has to do with frames. Actually, I have used this function "**buildModelFromUrdf**" to import **UR5e**  model. It works totally fine with me but the thing is: it specifies the frames totally different from what it shall be. For example, the first frame is specified automatically the "**universe**", but it shall be "**shoulder_pan_joint**" and the last frame is specified as "**wrist_3_joint**"  where it should be "**wrist_3_link-tool0_fixed_joint**". Therefore, based on these frames, The computation of Jacobian will result differently from what I have approached analytically!

Here is the code

```
model    = pin.buildModelFromUrdf(urdf_filename)
print('model name: ' + model.name)

# Create model and data
data = model.createData()

# Set bounds (by default they are undefinded for a the Simple Humanoid model)

model.lowerPositionLimit = -np.ones((model.nq,1))
model.upperPositionLimit = np.ones((model.nq,1))

# q = pin.randomConfiguration(model) # joint configuration
q = np.array([-2.93297704, -1.91732752,  1.66140008, -1.31517303, -1.57547641,
        0.21511427])

q_dot = np.random.rand(model.nv,1) # joint velocity
q_dotdot = np.random.rand(model.nv,1) # joint acceleration
f_ext = np.random.rand(model.nv,1) # joint acceleration

# pin.computeABADerivatives(model,data,q,q_dot,q_dotdot)
# pin.aba(model,data,q,v,a)
# Retrieve the derivatives in data
C = pin.computeCoriolisMatrix(model, data, q, q_dot)
pin.nonLinearEffects(model, data, q, q_dot)
G = pin.computeGeneralizedGravity(model, data, q)
pin.computeMinverse(model, data, q)
M = pin.crba(model, data, q)
J = pin.computeJointJacobians(model,data,q)

tau_crba =  M*q_dotdot + C*q_dot + G - J.T*f_ext
tau_crba_ = pin.crba(model, data, q)
```

jcarpent, 2023-12-05:
Your code is not consistent. Could you write down what you expect to do mathematically speaking?

jorisv, 2024-01-15:
Hello @abuibaid,

Did the last comment solve your issue ?

abuibaid, 2024-02-01:
@jorisv Thank you so much for your feedback. I understand it right now!

closed 2024-02-01 by abuibaid