stack-of-tasks/pinocchio #2130: pinocchio-preview3 compilation fails with errors related to casadi and eigenpy

EpicDuckPotato, 2024-01-11:
I want to use the pinocchio casadi python bindings. I cloned https://github.com/stack-of-tasks/pinocchio/tree/pinocchio3-preview, then ran the following sequence from within the pinocchio directory:
```cmd
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/openrobots -DBUILD_WITH_CASADI_SUPPORT=ON -DBUILD_WITH_AUTODIFF_SUPPORT=ON -DBUILD_WITH_COLLISION_SUPPORT=ON -DBUILD_WITH_CODEGEN_SUPPORT=ON -DBUILD_WITH_OPENMP_SUPPORT=ON
```

I get the following errors. I've installed the latest release of eigenpy (3.2) from source. I've also installed the latest release of casadi (3.6.4) from source. Any ideas what the issue could be?

```cmd
/opt/openrobots/include/eigenpy/eigen-from-python.hpp:35:8: error: ambiguous template instantiation for ‘struct eigenpy::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>, Eigen::MatrixBase<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1> > >’
   35 | struct expected_pytype_for_arg<
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
   36 |     Eigen::Matrix<Scalar, Rows, Cols, Options, MaxRows, MaxCols> >
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/openrobots/include/eigenpy/eigen-from-python.hpp:20:8: note: candidates are: ‘template<class MatType> struct eigenpy::expected_pytype_for_arg<MatType, Eigen::MatrixBase<Derived> > [with MatType = Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>]’
   20 | struct expected_pytype_for_arg<MatType, Eigen::MatrixBase<MatType> > {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/context.hpp:16,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/fwd.hpp:9,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/algorithm/algorithms.hpp:8,
                 from /home/noopygbhat/pinocchio/bindings/python/algorithm/expose-com.cpp:5:
/home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/context/casadi.hpp:64:8: note:                 ‘template<class CasadiScalar, int Rows, int Cols, int Options, int MaxRows, int MaxCols> struct eigenpy::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<T>, Rows, Cols, Options, MaxRows, MaxCols> > [with CasadiScalar = casadi::SXElem; int Rows = 3; int Cols = 1; int Options = 0; int MaxRows = 3; int MaxCols = 1]’
   64 | struct expected_pytype_for_arg<Eigen::Matrix<::casadi::Matrix<CasadiScalar>, Rows, Cols, Options, MaxRows, MaxCols> >
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/openrobots/include/eigenpy/details.hpp:11,
                 from /opt/openrobots/include/eigenpy/eigenpy.hpp:72,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/context/casadi.hpp:17,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/context.hpp:16,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/fwd.hpp:9,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/algorithm/algorithms.hpp:8,
                 from /home/noopygbhat/pinocchio/bindings/python/algorithm/expose-com.cpp:5:
/opt/openrobots/include/eigenpy/eigen-from-python.hpp:35:8: error: invalid use of incomplete type ‘struct eigenpy::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>, Eigen::MatrixBase<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1> > >’
   35 | struct expected_pytype_for_arg<
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
   36 |     Eigen::Matrix<Scalar, Rows, Cols, Options, MaxRows, MaxCols> >
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/openrobots/include/eigenpy/eigen-from-python.hpp:17:8: note: declaration of ‘struct eigenpy::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>, Eigen::MatrixBase<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1> > >’
   17 | struct expected_pytype_for_arg {};
      |        ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/boost/preprocessor/iteration/detail/iter/forward1.hpp:72,
                 from /usr/include/boost/python/detail/signature.hpp:42,
                 from /usr/include/boost/python/detail/caller.hpp:17,
                 from /usr/include/boost/python/object/function_handle.hpp:8,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:19,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from /opt/openrobots/include/eigenpy/fwd.hpp:73,
                 from /opt/openrobots/include/eigenpy/eigen-typedef.hpp:8,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/context/generic.hpp:9,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/context/casadi.hpp:11,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/context.hpp:16,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/fwd.hpp:9,
                 from /home/noopygbhat/pinocchio/build/include/pinocchio/bindings/python/algorithm/algorithms.hpp:8,
                 from /home/noopygbhat/pinocchio/bindings/python/algorithm/expose-com.cpp:5:
/usr/include/boost/preprocessor/iteration/detail/local.hpp: In instantiation of ‘static const boost::python::detail::signature_element* boost::python::detail::signature_arity<5>::impl<Sig>::elements() [with Sig = boost::mpl::vector6<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>, const pinocchio::ModelTpl<casadi::Matrix<casadi::SXElem>, 0>&, pinocchio::DataTpl<casadi::Matrix<casadi::SXElem>, 0>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&>]’:
/usr/include/boost/python/detail/caller.hpp:247:77:   required from ‘static boost::python::detail::py_func_sig_info boost::python::detail::caller_arity<5>::impl<F, Policies, Sig>::signature() [with F = Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1> (*)(const pinocchio::ModelTpl<casadi::Matrix<casadi::SXElem>, 0>&, pinocchio::DataTpl<casadi::Matrix<casadi::SXElem>, 0>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&); Policies = boost::python::return_value_policy<boost::python::return_by_value>; Sig = boost::mpl::vector6<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>, const pinocchio::ModelTpl<casadi::Matrix<casadi::SXElem>, 0>&, pinocchio::DataTpl<casadi::Matrix<casadi::SXElem>, 0>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&>]’
/usr/include/boost/python/object/py_function.hpp:48:34:   required from ‘boost::python::detail::py_func_sig_info boost::python::objects::caller_py_function_impl<Caller>::signature() const [with Caller = boost::python::detail::caller<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1> (*)(const pinocchio::ModelTpl<casadi::Matrix<casadi::SXElem>, 0>&, pinocchio::DataTpl<casadi::Matrix<casadi::SXElem>, 0>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&), boost::python::return_value_policy<boost::python::return_by_value>, boost::mpl::vector6<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, 1>, const pinocchio::ModelTpl<casadi::Matrix<casadi::SXElem>, 0>&, pinocchio::DataTpl<casadi::Matrix<casadi::SXElem>, 0>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&> >]’
/usr/include/boost/python/object/py_function.hpp:46:46:   required from here
[... cut ...] rix<casadi::SXElem>, 0>&, pinocchio::DataTpl<casadi::Matrix<casadi::SXElem>, 0>&, const Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1>&, long unsigned int> >]’
/usr/include/boost/python/object/py_function.hpp:46:46:   required from here
/usr/include/boost/preprocessor/iteration/detail/local.hpp:34:9: error: ‘get_pytype’ is not a member of ‘boost::python::converter::expected_pytype_for_arg<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 3, -1, 0> >’
   34 |         BOOST_PP_LOCAL_MACRO(0)
```

jcarpent, 2024-01-11:
@EpicDuckPotato Thanks for reporting this issue. Do you only the Python bindings?

EpicDuckPotato, 2024-01-11:
I only need the python bindings for my current application. I saw that I can get the pinocchio casadi bindings via conda, but I was hesitant to try this just because I currently use pip, and I was a bit worried using conda would cause additional headaches. Is conda the method you'd suggest?

jcarpent, 2024-01-11:
Yes, this is the best way to isolate your systems and deploy your programs on clusters.
You should definitely try it, and you will never come back to pip (even if you can use pip in conda envs)

closed 2024-01-12 by jcarpent