stack-of-tasks/pinocchio #2142: Installing via Linux fails from today (2024.01.31.)

PetoAdam, 2024-01-31:
## Bug description

When I try to install the package via the documentation at: https://stack-of-tasks.github.io/pinocchio/download.html, the installation fails, even though it worked perfectly yesterday.

## Expected behavior

The package should be installed correctly.

## Reproduction steps

I am installing the package to the default vscode image for now with this setup:

### Dockerfile (comment #-s taken out due to Markdown syntax):
Use the official VS Code Remote - Containers image
FROM mcr.microsoft.com/vscode/devcontainers/base:0-buster

Install required dependencies
RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y install --no-install-recommends lsb-release curl \
    && rm -rf /var/lib/apt/lists/*

Copy and run the installation script to the container
COPY scripts/install-pinocchio.sh /workspaces/scripts/install-pinocchio.sh
RUN chmod +x /workspaces/scripts/install-pinocchio.sh

Copy workspace folder to the container
COPY PhysicsService /workspaces/PhysicsService
RUN chown -R vscode /workspaces/PhysicsService

Switch back to non-root to install packages
USER vscode

Expose the environment variables
ENV PATH=/opt/openrobots/bin:$PATH
ENV PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
ENV LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
ENV PYTHONPATH=/opt/openrobots/lib/python3.10/site-packages:$PYTHONPATH
ENV CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH

### .devcontainer.json:
{
    "name": "CppPinocchioDevContainer",
    "dockerFile": "Dockerfile",
    "extensions": ["ms-vscode.cpptools"],
    "settings": {
      "terminal.integrated.shell.linux": "/bin/bash"
    },
    "forwardPorts": [22],
    "postCreateCommand": "/workspaces/scripts/install-pinocchio.sh",
    "workspaceMount": "source=remote-container-home,target=/workspaces/PhysicsService",
    "shutdownAction": "stopCompose",
    "remoteUser": "vscode"
  }
  
  
  ### and here is the install-pinocchio.sh script:
  #!/bin/bash

Run apt update to fetch package descriptions
sudo apt update -qqy

Add robotpkg as a source repository
sudo apt install -qqy lsb-release curl
sudo mkdir -p /etc/apt/keyrings
curl http://robotpkg.openrobots.org/packages/debian/robotpkg.asc \
     | sudo tee /etc/apt/keyrings/robotpkg.asc
echo "deb [arch=amd64 signed-by=/etc/apt/keyrings/robotpkg.asc] http://robotpkg.openrobots.org/packages/debian/pub $(lsb_release -cs) robotpkg" \
     | sudo tee /etc/apt/sources.list.d/robotpkg.list

Run apt update to fetch package descriptions
sudo apt update -qqy

Install Pinocchio and its dependencies
sudo apt install -qqy robotpkg-py3*-pinocchio

Configure environment variables
echo 'export PATH=/opt/openrobots/bin:$PATH' >> /home/vscode/.bashrc
echo 'export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH' >> /home/vscode/.bashrc
echo 'export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH' >> /home/vscode/.bashrc
echo 'export PYTHONPATH=/opt/openrobots/lib/python3.10/site-packages:$PYTHONPATH' >> /home/vscode/.bashrc
echo 'export CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH' >> /home/vscode/.bashrc

### This script seems to fail with the issue:
deb [arch=amd64 signed-by=/etc/apt/keyrings/robotpkg.asc] http://robotpkg.openrobots.org/packages/debian/pub buster robotpkg
32 packages can be upgraded. Run 'apt list --upgradable' to see them.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-py37-pinocchio : Depends: robotpkg-py37-hpp-fcl (= 1.7.3r1) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.8.1) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 2.3.6) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.4) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 2.4.0) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.0) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.5.4) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.2) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 2.3.5) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.4.4) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 2.2.0) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.8) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 2.3.0) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.5) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.8.0) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.3) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.4r2) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.4r1) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.4.4r1) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.4.2) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.1r1) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.7.1) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.6.0) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.4.5) but it is not going to be installed or
                                    robotpkg-py37-hpp-fcl (= 1.3.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

## Additional context

The same script was working perfectly yesterday, but today I rebuilt the container and met this issue.

### System

- OS: Docker (vscode image)
- Pinocchio version: newest

nim65s, 2024-01-31:
Thanks for the report. We lack proper testing on Debian for now, and your Docker image is based on Debian. I'll investigate.

nim65s, 2024-01-31:
Oh, This is debian buster, which is currently "oldoldstable", could you try to upgrade to bullseye or bookworm ?
` FROM mcr.microsoft.com/vscode/devcontainers/base:0-buster` → `FROM mcr.microsoft.com/devcontainers/base:bookworm` ?

nim65s, 2024-01-31:
Also there are mentions of both python 3.7 and python 3.10 in your setup. Make sure to use only one: the one from your debian system.

nim65s, 2024-01-31:
Actually, we forgot to release bookworm… so please try:
````diff
-mcr.microsoft.com/vscode/devcontainers/base:0-buster
+FROM mcr.microsoft.com/devcontainers/base:1.0-bullseye
```

PetoAdam, 2024-01-31:
It works! I was already planning on upgrading to bullseye, so this really makes me do it. :) 
Thanks for the heads-up regarding the python version, I dont't use it inside the container, but at least that is a possible debug issue solved 👍 
Thanks a lot!

closed 2024-01-31 by jcarpent