stack-of-tasks/pinocchio #2145: Compilation Issues with Pinocchio Headers on Windows Using MSVC and Clang

ShadowDawg, 2024-02-01:
I'm encountering problems when attempting to compile a simple C++ project with Pinocchio headers on Windows. The project fails to compile with both MSVC and Clang compilers, presenting different issues.

**MSVC Compiler:** Results in over 100 errors before compilation terminates.
**Clang Compiler:** While the cmake command executes without issues, running make leads to the error: 
```
clang++: error: no such file or directory: '/bigobj'.
```

The Pinocchio documentation only mentions using the GCC compiler but due to project constraints I have to use either MSVC or Clang.
Are there any specific instructions or updates to support compilation with MSVC or Clang on Windows? It would be really helpful to me.

Thanks in advance!

jcarpent, 2024-02-01:
Could you share your log files?
Could you share your command entries?

ShadowDawg, 2024-02-01:
**Using MSVC:**
All commands have been run in the visual studio prompt in an activated conda environment containing the pinocchio package.

Command entries:
```
>> cmake -G "Visual Studio 17 2022" ..
>> cmake --build . --config Release
```

Output:
```
Build started at 18:27...
1>------ Build started: Project: kinematics, Configuration: Release x64 ------
1>cl : command line  warning D9002: ignoring unknown option '/EH;'
1>kinematics.cpp
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(70,5): error C2988: unrecognizable template declaration/definition
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(70,5): error C2059: syntax error: 'const'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(70,5): error C2059: syntax error: ')'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(70,5): error C2143: syntax error: missing ';' before '{'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(70,5): error C2447: '{': missing function header (old-style formal list?)
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(71,5): error C2988: unrecognizable template declaration/definition
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(71,5): error C2059: syntax error: 'const'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(71,5): error C2059: syntax error: ')'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(71,5): error C2143: syntax error: missing ';' before '{'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\math\fwd.hpp(71,5): error C2447: '{': missing function header (old-style formal list?)
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hpp(284,50): error C2589: '(': illegal token on right side of '::'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hpp(284,17): error C2062: type 'unknown-type' unexpected
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hpp(284,55): error C2612: trailing ')' illegal in base/member initializer list
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hxx(15,3): error C2888: 'pinocchio::CollisionPair::CollisionPair(void)': symbol cannot be defined within namespace 'pinocchio'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hxx(19,3): error C2888: 'pinocchio::CollisionPair::CollisionPair(const pinocchio::GeomIndex,const pinocchio::GeomIndex)': symbol cannot be defined within namespace 'pinocchio'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hxx(25,3): error C2888: 'bool pinocchio::CollisionPair::operator ==(const pinocchio::CollisionPair &) const': symbol cannot be defined within namespace 'pinocchio'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hxx(31,3): error C2888: 'bool pinocchio::CollisionPair::operator !=(const pinocchio::CollisionPair &) const': symbol cannot be defined within namespace 'pinocchio'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hxx(32,20): error C2264: 'pinocchio::CollisionPair::operator ==': error in function definition or declaration; function not called
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hxx(32,20): error C2088: '==': illegal for struct
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio\multibody\fcl.hxx(36,3): error C2888: 'void pinocchio::CollisionPair::disp(std::ostream &) const': symbol cannot be defined within namespace 'pinocchio'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\include\list(27,31): error C2061: syntax error: identifier 'bidirectional_iterator_tag'
1>(compiling source file '../../../src/kinematics/kinematics.cpp')
1>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\include\list(27,31):
1>the template instantiation context (the oldest one first) is
[... cut ...]
find_package(Eigen3 REQUIRED)
find_package(pinocchio REQUIRED)
# Set kinematics library source files
set(KINEMATICS_SOURCES
	kinematics.cpp
)
#add_library(${PROJECT_NAME} STATIC ${KINEMATICS_SOURCES})
add_executable(kinematics kinematics.cpp)
target_include_directories(${PROJECT_NAME} PUBLIC 
						   ${synapse_SOURCE_DIR}/include 
)
target_link_libraries(${PROJECT_NAME} PUBLIC Eigen3::Eigen pinocchio::pinocchio)

```

Hope this clears things up and I'd be happy to provide more details if I missed anything.

ShadowDawg, 2024-02-01:
This is my conda list:
```
# platform: win-64
aom=3.7.1=h63175ca_0
assimp=5.3.1=h81f0834_2
azure-core-cpp=1.11.0=h249a519_0
azure-storage-blobs-cpp=12.10.0=h91493d7_0
azure-storage-common-cpp=12.5.0=h91493d7_2
blosc=1.21.5=hdccc3a2_0
bullet-cpp=3.25=h2ab9e98_2
bzip2=1.0.8=hcfcfb64_5
c-ares=1.25.0=hcfcfb64_0
ca-certificates=2023.11.17=h56e8100_0
cairo=1.18.0=h1fef639_0
cfitsio=4.3.1=h9b0cee5_0
cmake=3.28.1=hf0feee3_0
console_bridge=1.0.2=h5362a0b_1
cppzmq=4.10.0=h42135b4_1
dartsim=6.13.1=hf6eeb0b_2
dav1d=1.2.1=hcfcfb64_0
dirent=1.21=0
dlfcn-win32=1.3.0=h0e60522_0
eigen=3.4.0=h91493d7_0
eigenpy=3.3.0=py312h0bfd76e_0
expat=2.5.0=h63175ca_1
fcl=0.7.0=he22821c_4
ffmpeg=6.1.1=gpl_hadb5375_101
flann=1.9.2=h23e6bae_0
fmt=10.2.1=h181d51b_0
font-ttf-dejavu-sans-mono=2.37=hab24e00_0
font-ttf-inconsolata=3.000=h77eed37_0
font-ttf-source-code-pro=2.038=h77eed37_0
font-ttf-ubuntu=0.83=h77eed37_1
fontconfig=2.14.2=hbde0cde_0
fonts-conda-ecosystem=1=0
fonts-conda-forge=1=0
freeglut=3.2.2=h63175ca_2
freeimage=3.18.0=h609497f_18
freetype=2.12.1=hdaf720e_2
freexl=2.0.0=h8276f4a_0
fribidi=1.0.10=h8d14728_0
gazebo=11.14.0=he2f2283_6
geos=3.12.1=h1537add_0
geotiff=1.7.1=hbf5ca3a_15
getopt-win32=0.1=hcfcfb64_1
gettext=0.21.1=h5728263_0
glib=2.78.3=h12be248_0
glib-tools=2.78.3=h12be248_0
graphite2=1.3.13=1000
graphviz=9.0.0=h51cb2cd_1
gst-plugins-base=1.22.8=h001b923_1
gstreamer=1.22.8=hb4038d2_1
gts=0.7.6=h6b5321d_4
harfbuzz=8.3.0=h7ab893a_0
hdf4=4.2.15=h5557f11_7
hdf5=1.14.3=nompi_h73e8ff5_100
hpp-fcl=2.4.1=py312hf774def_0
icu=73.2=h63175ca_0
imath=3.1.9=h12be248_0
intel-openmp=2024.0.0=h57928b3_49840
jsoncpp=1.9.5=h2d74725_1
jxrlib=1.1=hcfcfb64_3
kealib=1.5.3=hd248416_0
krb5=1.21.2=heb0366b_0
lcms2=2.16=h67d730c_0
lerc=4.0.0=h63175ca_0
libabseil=20230802.1=cxx17_h63175ca_0
libaec=1.1.2=h63175ca_1
libarchive=3.7.2=h313118b_1
libblas=3.9.0=21_win64_mkl
libboost=1.82.0=h65993cd_6
libboost-devel=1.82.0=h91493d7_6
libboost-headers=1.82.0=h57928b3_6
libboost-python=1.82.0=py312h4f1204c_6
libboost-python-devel=1.82.0=py312hd42ba9a_6
libcblas=3.9.0=21_win64_mkl
libccd-double=2.1=h63175ca_3
libclang=15.0.7=default_hde6756a_4
libclang13=15.0.7=default_h85b4d89_4
libcrc32c=1.1.2=h0e60522_0
libcurl=8.5.0=hd5e4a3a_0
libdeflate=1.19=hcfcfb64_0
libexpat=2.5.0=h63175ca_1
libffi=3.4.2=h8ffe710_5
libflang=5.0.0=h6538335_20180525
libgd=2.3.3=h312136b_9
libgdal=3.8.3=h576f4c1_0
libglib=2.78.3=h16e383f_0
libgoogle-cloud=2.12.0=h39f2fc6_4
libgrpc=1.59.3=h5bbd4a7_0
libhwloc=2.9.3=default_haede6df_1009
libiconv=1.17=hcfcfb64_2
libignition-cmake2=2.16.0=h63175ca_1
libignition-common3=3.15.1=hef572a9_3
libignition-fuel-tools4=4.6.0=he14a2b9_6
libignition-math6=6.15.1=py312haf63811_1
libignition-msgs5=5.11.0=h8c01862_5
libignition-tools1=1.5.0=hf5993cb_3
libignition-transport8=8.4.0=hbed1618_7
libjpeg-turbo=3.0.0=hcfcfb64_1
libkml=1.3.0=haf3e7a6_1018
liblapack=3.9.0=21_win64_mkl
libnetcdf=4.9.2=nompi_h07c049d_113
libode=0.16.2=h53d5487_12
libogg=1.3.4=h8ffe710_1
libopenblas=0.3.26=pthreads_hc140b1d_0
libopus=1.3.1=h8ffe710_1
libpng=1.6.39=h19919ed_0
libpq=16.1=hdb24f17_7
libprotobuf=4.24.4=hb8276f3_0
libraw=0.21.1=h5557f11_2
libre2-11=2023.06.02=h8c5ae5e_0
librttopo=1.1.0=h94c4f80_15
libsdformat=9.8.0=hbbad600_6
libsodium=1.0.18=h8d14728_1
libspatialite=5.1.0=hf2f0abc_4
libsqlite=3.44.2=hcfcfb64_0
libssh2=1.11.0=h7dfc565_0
libtiff=4.6.0=h6e2ebb7_2
libusb=1.0.26=h8ffe710_100
libuv=1.44.2=hcfcfb64_1
libvorbis=1.3.7=h0e60522_0
libwebp=1.3.2=hcfcfb64_1
libwebp-base=1.3.2=hcfcfb64_0
libxcb=1.15=hcd874cb_0
libxml2=2.12.4=hc3477c8_1
libzip=1.10.1=h1d365fa_3
libzlib=1.2.13=hcfcfb64_5
llvm-meta=5.0.0=0
lz4-c=1.9.4=hcfcfb64_0
lzo=2.10=he774522_1000
m2w64-gcc-libgfortran=5.3.0=6
m2w64-gcc-libs=5.3.0=7
m2w64-gcc-libs-core=5.3.0=7
m2w64-gmp=6.1.0=2
m2w64-libwinpthread-git=5.0.0.4634.697f757=2
make=4.3=h3d2af85_1
minizip=4.0.4=h5bed578_0
mkl=2024.0.0=h66d3029_49657
msys2-conda-epoch=20160418=1
numpy=1.26.3=py312h8753938_0
octomap=1.9.8=h91493d7_0
ogre=1.10.12=hc646683_16
openal-soft=1.23.1=h91493d7_0
openblas=0.3.26=pthreads_h3721920_0
openexr=3.2.1=h5fba010_0
openh264=2.4.0=h63175ca_0
openjpeg=2.5.0=h3d672ee_3
openmp=5.0.0=vc14_1
openssl=3.2.0=hcfcfb64_1
pango=1.50.14=h07c897b_2
pcre2=10.42=h17e33f8_0
pinocchio=2.7.0=py312h24f9807_0
pip=23.3.2=pyhd8ed1ab_0
pixman=0.43.0=h63175ca_0
pkg-config=0.29.2=h2bf4dc2_1008
poppler=23.12.0=hc2f3c52_0
poppler-data=0.4.12=hd8ed1ab_0
postgresql=16.1=h1beaf6b_7
proj=9.3.1=he13c7e8_0
pthread-stubs=0.4=hcd874cb_1001
pthreads-win32=2.9.1=hfa6e2cd_3
pugixml=1.14=h63175ca_0
pybind11-abi=4=hd8ed1ab_3
python=3.12.1=h2628c8c_1_cpython
python_abi=3.12=4_cp312
qhull=2020.2=h70d2c02_2
qt-main=5.15.8=h9e85ed6_18
qwt=6.2.0=h07be427_6
re2=2023.06.02=hcbb65ff_0
ruby=3.2.2=h20ad4f3_1
sdl2=2.28.5=h63175ca_0
setuptools=69.0.3=pyhd8ed1ab_0
simbody=3.7=hade3207_3
snappy=1.1.10=hfb803bf_0
spdlog=1.12.0=h64d2f7d_2
sqlite=3.44.2=hcfcfb64_0
svt-av1=1.8.0=h63175ca_0
tbb=2021.11.0=h91493d7_0
tbb-devel=2021.11.0=h91493d7_0
tiledb=2.19.0=h8e52ccb_0
tiny-process-library=2.0.4=h63175ca_1
tinyxml=2.6.2=h2d74725_2
tinyxml2=10.0.0=h63175ca_0
tk=8.6.13=h5226925_1
tzdata=2023d=h0c530f3_0
ucrt=10.0.22621.0=h57928b3_0
urdfdom=4.0.0=h0b06ded_1
urdfdom_headers=1.1.1=h91493d7_0
uriparser=0.9.7=h1537add_1
vc=14.3=hcf57466_18
vc14_runtime=14.38.33130=h82b7239_18
vs2015_runtime=14.38.33130=hcb4865c_18
wheel=0.42.0=pyhd8ed1ab_0
x264=1!164.3095=h8ffe710_2
x265=3.5=h2d74725_3
xerces-c=3.2.5=h63175ca_0
xorg-kbproto=1.0.7=hcd874cb_1002
xorg-libice=1.1.1=hcd874cb_0
xorg-libsm=1.2.4=hcd874cb_0
xorg-libx11=1.8.7=hefa74cf_0
xorg-libxau=1.0.11=hcd874cb_0
xorg-libxdmcp=1.1.3=hcd874cb_0
xorg-libxext=1.3.4=hcd874cb_2
xorg-libxpm=3.5.17=hcd874cb_0
xorg-libxt=1.3.0=hcd874cb_1
xorg-xextproto=7.3.0=hcd874cb_1003
xorg-xproto=7.0.31=hcd874cb_1007
xz=5.2.6=h8d14728_0
yaml=0.2.5=h8ffe710_2
zeromq=4.3.5=h63175ca_0
zlib=1.2.13=hcfcfb64_5
zstd=1.5.5=h12be248_0
zziplib=0.13.69=h1d00b33_1
```

traversaro, 2024-02-01:
Nothing strage seems to be in the env variable. Just to try, can you try to install `vs2022_win-64` and `ninja` and try to configure the project with `cmake -GNinja ..` instead of `cmake -G "Visual Studio 17 2022" ..` and see if you have the same problem?

ShadowDawg, 2024-02-02:
Thanks for the suggestions. After installing `vs2022_win-64` and running `cmake -GNinja ..` followed by `ninja`, the same errors seem to occur. Attaching the output below.
```
[1/4] Building CXX object src\kinematics\CMakeFiles\kinematics.dir\kinematics.cpp.obj
FAILED: src/kinematics/CMakeFiles/kinematics.dir/kinematics.cpp.obj
C:\PROGRA~1\MIB055~1\2022\COMMUN~1\VC\Tools\MSVC\1438~1.331\bin\Hostx64\x64\cl.exe  /nologo /TP -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DHPP_FCL_HAS_OCTOMAP -DHPP_FCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=8 -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -IC:\Users\shado\miniconda3\envs\last\Library\include\gazebo-11 -IC:\Users\shado\miniconda3\envs\last\Library\include\sdformat-9.8 -IC:\Users\shado\miniconda3\envs\last\Library\include\ignition\math6 -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\Bites -IC:\Users\shado\miniconda3\envs\last\Library\include\SDL2 -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\HLMS -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\MeshLodGenerator -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\Overlay -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\Paging -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\Property -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\RTShaderSystem -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\Terrain -IC:\Users\shado\miniconda3\envs\last\Library\include\OGRE\Volume -IC:\Users\shado\miniconda3\envs\last\Library\include\ignition\transport8 -IC:\Users\shado\miniconda3\envs\last\Library\include\ignition\msgs5 -IC:\Users\shado\miniconda3\envs\last\Library\include\ignition\common3 -IC:\Users\shado\miniconda3\envs\last\Library\include\ignition\fuel_tools4 -IC:\Users\shado\OneDrive\Desktop\work\synapse-gazebo\include -I\include -external:IC:\Users\shado\miniconda3\envs\last\Library\include -external:IC:\Users\shado\miniconda3\envs\last\Library\include\eigen3 -external:W0 /DWIN32 /D_WINDOWS /W3 /GR /EHsc; /Zc:__cplusplus /MDd /Zi /Ob0 /Od /RTC1 -std:c++14 /bigobj /showIncludes /Fosrc\kinematics\CMakeFiles\kinematics.dir\kinematics.cpp.obj /Fdsrc\kinematics\CMakeFiles\kinematics.dir\ /FS -c C:\Users\shado\OneDrive\Desktop\work\synapse-gazebo\src\kinematics\kinematics.cpp
cl : Command line warning D9002 : ignoring unknown option '/EH;'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(70): error C2988: unrecognizable template declaration/definition
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(70): error C2059: syntax error: 'const'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(70): error C2059: syntax error: ')'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(70): error C2143: syntax error: missing ';' before '{'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(70): error C2447: '{': missing function header (old-style formal list?)
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(71): error C2988: unrecognizable template declaration/definition
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(71): error C2059: syntax error: 'const'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(71): error C2059: syntax error: ')'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(71): error C2143: syntax error: missing ';' before '{'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/math/fwd.hpp(71): error C2447: '{': missing function header (old-style formal list?)
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/multibody/fcl.hpp(284): error C2589: '(': illegal token on right side of '::'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/multibody/fcl.hpp(284): error C2062: type 'unknown-type' unexpected
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/multibody/fcl.hpp(284): error C2612: trailing ')' illegal in base/member initializer list
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/multibody/fcl.hxx(15): error C2888: 'pinocchio::CollisionPair::CollisionPair(void)': symbol cannot be defined within namespace 'pinocchio'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/multibody/fcl.hxx(19): error C2888: 'pinocchio::CollisionPair::CollisionPair(const pinocchio::GeomIndex,const pinocchio::GeomIndex)': symbol cannot be defined within namespace 'pinocchio'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/multibody/fcl.hxx(25): error C2888: 'bool pinocchio::CollisionPair::operator ==(const pinocchio::CollisionPair &) const': symbol cannot be defined within namespace 'pinocchio'
C:\Users\shado\miniconda3\envs\last\Library\include\pinocchio/multibody/fcl.hxx(31): error C2888: 'bool pinocchio::CollisionPair::operator !=(const pinocchio::CollisionPair &) const': symbol cannot be defined within namespace 'pinocchio'
[... cut ...]
[2/4] Building CXX object CMakeFiles\ur5_plugin.dir\src\ro...ersal_robotics\ur5_gazebo\gazebo_plugin\ur5_plugin.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '/EH;'
Please define _WIN32_WINNT or _WIN32_WINDOWS appropriately. For example:
- add -D_WIN32_WINNT=0x0601 to the compiler command line; or
- add _WIN32_WINNT=0x0601 to your project's Preprocessor Definitions.
Assuming _WIN32_WINNT=0x0601 (i.e. Windows 7 target).
ninja: build stopped: subcommand failed.
```

Is there anything else I could try?

traversaro, 2024-02-02:
> Is there anything else I could try?

You can actually try clang by installing the clang package, and setting:
~~~
set "CC=clang-cl.exe"
set "CXX=clang-cl.exe"
~~~

and then configure with Ninja on a clean build. From the CI it seems that also `cl.exe` should be supported, but for sure clang should be a safer bet.

ShadowDawg, 2024-02-02:
Hi @jorisv . 
I followed your instructions and I'm glad to say that it works now and I was able to run the example from the docs too.
Thank you so much jorisv, @traversaro and @jcarpent for your time and your help!

closed 2024-02-02 by ShadowDawg