stack-of-tasks/pinocchio #2148: Issue with install Pinocchio on Jetson Orin

lwygzh, 2024-02-01:
## Bug description

Gives an error when install pinocchio using `pip install pin` inside an Isaac ROS docker container deployed on a NVIDIA Jetson Orin.

## Expected behavior

Installs pinocchio successfully.

## Reproduction steps

1. Follow the steps [here](https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html) to install the Isaac ROS docker
2. Run `run_dev.sh` following the steps [here](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_common/index.html#isaac-ros-docker-development-environment) to run the docker container
3. run `pip install pin` inside the container.

### Docker image

See above

## Additional context

This is the error message:

```bash
python3 -m pip install pin==2.6.21 proxsuite
Collecting pin==2.6.21
  Downloading pin-2.6.21.tar.gz (61.8 MB)
     |████████████████████████████████| 61.8 MB 15.2 MB/s 
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-wy7o9b6f/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'cmeel-boost ~= 1.83.0' 'cmeel-urdfdom[build] >= 3.1.0.3, < 4' 'cmeel[build]' 'hpp-fcl[build] >= 2.3.4, < 4'
       cwd: None
  Complete output (296 lines):
  Collecting cmeel-boost~=1.83.0
    Using cached cmeel_boost-1.83.0.tar.gz (4.0 kB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: still running...
      Preparing wheel metadata: still running...
      Preparing wheel metadata: still running...
      Preparing wheel metadata: still running...
      Preparing wheel metadata: finished with status 'done'
  Collecting cmeel-urdfdom[build]<4,>=3.1.0.3
    Downloading cmeel_urdfdom-3.1.1.1.tar.gz (292 kB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting cmeel[build]
    Using cached cmeel-0.53.3-py3-none-any.whl (20 kB)
  Collecting hpp-fcl[build]<4,>=2.3.4
    Downloading hpp_fcl-2.4.1.tar.gz (3.2 MB)
    Installing build dependencies: started
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: still running...
    Installing build dependencies: finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-1tiji5hd/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'cmeel-assimp >= 5.3.1, < 6' 'cmeel-boost ~= 1.83.0' 'cmeel-octomap >= 1.9.8.2, < 2' 'cmeel-qhull >= 8.0.2.1, < 8.0.3' 'cmeel[build]' 'eigenpy[build] >= 3.1, < 4'
[... 242 lines cut ...]
            return _get_wheel_metadata_from_wheel(backend, metadata_directory,
          File "/tmp/tmpfe2m5gwd", line 159, in _get_wheel_metadata_from_wheel
            whl_basename = backend.build_wheel(metadata_directory, config_settings)
          File "/tmp/pip-build-env-ji3vfh3k/overlay/lib/python3.8/site-packages/cmeel/build.py", line 26, in build_wheel
            return build_impl(wheel_directory, editable=False)
          File "/tmp/pip-build-env-ji3vfh3k/overlay/lib/python3.8/site-packages/cmeel/impl.py", line 107, in build_impl
            launch_tests(True, run_tests and not run_tests_after_install, pyproject, build)
          File "/tmp/pip-build-env-ji3vfh3k/overlay/lib/python3.8/site-packages/cmeel/utils.py", line 231, in launch_tests
            check_call(cmd, env=test_env)
          File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
            raise CalledProcessError(retcode, cmd)
        subprocess.CalledProcessError: Command '['cmake', '--build', '/tmp/cmeel-m3uzoocm/bld', '-t', 'test']' returned non-zero exit status 2.
        ----------------------------------------
    ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmpfe2m5gwd prepare_metadata_for_build_wheel /tmp/tmptsnph3rn Check the logs for full command output.
    ----------------------------------------
  ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-1tiji5hd/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'cmeel-assimp >= 5.3.1, < 6' 'cmeel-boost ~= 1.83.0' 'cmeel-octomap >= 1.9.8.2, < 2' 'cmeel-qhull >= 8.0.2.1, < 8.0.3' 'cmeel[build]' 'eigenpy[build] >= 3.1, < 4' Check the logs for full command output.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-wy7o9b6f/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'cmeel-boost ~= 1.83.0' 'cmeel-urdfdom[build] >= 3.1.0.3, < 4' 'cmeel[build]' 'hpp-fcl[build] >= 2.3.4, < 4' Check the logs for full command output.

```

### System

- OS: [e.g. Ubuntu 20.04]
- Pinocchio version: [e.g. 2.7.0]

lvjonok, 2024-03-02:
It looks like you have an issue not directly related to pinocchio itself rather to its [dependency](https://github.com/cmake-wheel/cmeel-octomap).

I think the problem is that `cmeel-octomap` does not have pre-built package for your architecture and whilst trying to install you encounter some error.

nim65s, 2024-03-02:
cmeel-octomap should be available for python 3.8 on aarch64. I'll double check this on Monday.

nim65s, 2024-03-02:
@lwygzh : just to be sure, could you try again after a `python3 -m pip install -U --user pip` ?

lvjonok, 2024-03-02:
Yes, `cmeel-octomap` seem to contain `aarch64` built, may it be the case that it should be something different from `manylinux` ?
[image]

nim65s, 2024-03-02:
My guess is that the pip version is too old for manylinux_2_28. After a pip upgrade, it should be fine.

piliwilliam0306, 2024-10-03:
`My guess is that the pip version is too old for manylinux_2_28. After a pip upgrade, it should be fine.`
pip upgrade worked for me, thanks!