stack-of-tasks/pinocchio #2150: pin.constraintDynamics and data.lambda_c

MR-KNV, 2024-02-02:
Hello!
I would like to thank you and your team for the Pinocchio library. I used Pinocchio 2.99.0 from this conda package: [package](https://anaconda.org/olivier.roussel/pinocchio)
I have a model with closed kinematic loop and want to calculate dynamics of it. 
I checked in Meshcat viewer that **inverseConstraintKinematic** works well but Im not sure that vector of `data.lambda_c` is calculated correctly. I know exactly, that vector of `data.lambda_c` should be about 3-5 N by X axis, but I got an array with values that are close to zero.

I use this command for calculations:

`pin.initConstraintDynamics(model, data, constraint_model)`
`pin.rnea(model, data, q, vq, ddq)  % to calculate data.tau` 
`ddq= pin.constraintDynamics(model, data, q, vq, data.tau, constraint_model, constraint_data, prox_settings)  % update ddq`

Could you please tell me if my calculations are correct? Is there anywhere I can read the documentation for the pin.constraint Dynamics ()method? 

Thank you!

jorisv, 2024-02-08:
Hello @MR-KNV,

Can you give us a script to reproduce your issue ?

The method is described in this paper [Justin Carpentier 2021](https://inria.hal.science/hal-03271811/file/rss-proximal-and-sparse.pdf)

MR-KNV, 2024-02-08:
Yes, my script is here ([https://github.com/MR-KNV/pinocchio_test](https://github.com/MR-KNV/pinocchio_test))