stack-of-tasks/pinocchio #2152: Compatibility with older Boost versions on Windows

ShadowDawg, 2024-02-07:
An existing project of mine utilizes Boost 1.79, while Pinocchio requires a minimum version of 1.82 to build successfully. Integrating Pinocchio with my project is crucial, but the Boost versions are conflicting!

Is there a documented or supported way to use Pinocchio on Windows with Boost 1.79 or a similar older version? If not, are there any plans to add such compatibility in future releases?

jcarpent, 2024-02-07:
How do you install Pinocchio?
Pinocchio is not stick to a particular version of Boost.

ShadowDawg, 2024-02-07:
I've installed Pinocchio using Conda Forge. On building the first example in the docs (using msvc), it throws the following error:
```
CMake Error at C:/Program Files/CMake/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake: 230 (message): Could NOT find Boost: Found unsuitable version "1.79.0", but required is at
least "1.82" (found C:/Program Files/boost/boost_1_79_0, found components: system thread chrono atomic)
```
My CMakeLists.txt file for the test program is:
```
cmake_minimum_required(VERSION 3.8)
project(pinocchio_example)

set(CMAKE_CXX_STANDARD 11)

# Pinocchio must be build with this define on Windows to avoid conflict with the Windows min/max macro.
add_definitions("-DNOMINMAX")

find_package(Eigen3 REQUIRED)
find_package(pinocchio REQUIRED)

# Include directories for Eigen and Pinocchio
include_directories(${EIGEN3_INCLUDE_DIR} ${pinocchio_INCLUDE_DIRS})

# Define the executable for the kinematics example
add_executable(${PROJECT_NAME} overview-simple.cpp)

# Link Eigen and Pinocchio libraries
target_link_libraries(${PROJECT_NAME} Eigen3::Eigen ${pinocchio_LIBRARIES})
```
I assumed that Pinocchio requires a certain version of Boost since it seems to use some of its  components such as filesystem and system. 
Hope this provides the necesarry information.

ManifoldFR, 2024-02-07:
> I've installed Pinocchio using Conda Forge. On building the first example in the docs (using msvc), it throws the following error:
> ```
> CMake Error at C:/Program [...]

If you installed Pinocchio via conda it installed a version of Boost from conda also. What's probably happening there is that you are (or your CMake is) trying to build against your local boost in Program Files instead of the one that Pinocchio came (and was built and dynamically linked) with. CMake saw that and was not happy, hence your error.

If you want to use your local version of Boost, you ought to build Pinocchio (and its dependencies eigenpy and hppfcl) locally with that version of Boost so as to have consistent versions and linked symbols.

Another (less painful) option is to build against conda's Boost and properly tell CMake to look for Boost inside your conda environment. For this you should set the CMAKE_PREFIX_PATH environment variable to your conda environment path _before_ the first call to CMake, and do so for _every_ project you want to build against the conda-installed Pinocchio.

ManifoldFR, 2024-02-07:
I'll close this as it's not related to Pinocchio itself

closed 2024-02-07 by ManifoldFR