stack-of-tasks/pinocchio #2157: VS Tools installation bug

KamV1, 2024-02-22:
## Bug description

When I try to install Stable Cascade, I open a window for stable Cascade in Pinokio, and then click the download button. After that, I see the window with the installation requirereds and click install. After that, it started to install all the components in new window that were needed but on the last component VS tools it just did not work, I tried to install it multiple times, and in the installation window it showed that the installation succeeded, but then when I click ok I see again windows with all needed components and down there I see VS - Not installed.

## Expected behavior

A clear and concise description of what you expected to happen.

## Reproduction steps

Steps to reproduce the behavior:
1. Load to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Docker image

If this is an installation issue, a Docker image to reproduce it

```docker file

...
```

### Code

If this is a runtime issue, provide some minimal code to reproduce it:

```
...
```

## Additional context
Add any other context about the problem here.

### Screenshots
[image]
[image]

### System

- OS: [e.g. iOS]
- Pinocchio version: [e.g. 0.1.0]

fabinsch, 2024-02-22:
Hi @Punsihmentor, I think this here is the wrong repo. Maybe you are looking for: https://github.com/pinokiocomputer/pinokio ?

closed 2024-02-22 by fabinsch

Rhanderguimararaes, 2024-03-12:
It seems like it's a permission issue, but how to allow