stack-of-tasks/pinocchio #2159: ImportError: cannot import name 'casadi' from 'pinocchio'

lrchit, 2024-03-04:
## Bug description

I'm doing nmpc with casadi and acados. Recently I need some dynamics to build a model, such as  Centroid dynamics. So I tried to bind pinocchio with casadi using python. When I did cmake and make, everything went well. But when I tried python, something was wrong. 
After everything was installed, I ran python3 and tried "from pinocchio import casadi", at which errors occured as followings:
[image]

[image] 

### Code

Considering something needs eigenpy 2.7.10, I installed it in advanced.
I used following steps to build pinocchio with a virtual environment:

```
virtualenv env --python=/usr/bin/python3
source env/bin/activate
pip install cmake casadi compilers
pip install pinocchio
cmake . -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_WITH_CASADI_SUPPORT=ON
sudo make install
```
I tested pinocchio with my previous C++ project and that did well. As there was nothing wrong when compiling, so I don't know how to solve this. 
Is there anything wrong with the virtual environment or cmake procedure? Thanks in advance!

### System

- OS: ubuntu20.04
- Pinocchio version: pinocchio-preview3

jcarpent, 2024-03-04:
You can install pinocchio3 preview with modern features via conda directly.
I recommend using the procedure detailed here: https://github.com/agimus-project/winter-school-2023/tree/main/simulation/sim1_rigid_body

Best,
Justin

closed 2024-03-04 by jcarpent

lrchit, 2024-03-05:
> You can install pinocchio3 preview with modern features via conda directly. I recommend using the procedure detailed here: https://github.com/agimus-project/winter-school-2023/tree/main/simulation/sim1_rigid_body
> 
> Best, Justin

> Thanks a lot for reply. I followed https://github.com/agimus-project/winter-school-2023/tree/main/simulation/sim1_rigid_body just now with conda, but overcome the same error:
[image]

> I guess maybe this is caused by the python version? I use python3.8. Looking forward to your reply. Thanks again.
> Best wishes!

lrchit, 2024-03-05:
> You can install pinocchio3 preview with modern features via conda directly. I recommend using the procedure detailed here: https://github.com/agimus-project/winter-school-2023/tree/main/simulation/sim1_rigid_body
> 
> Best, Justin

Dear Justin, 
I followed https://github.com/agimus-project/winter-school-2023/tree/main/simulation/sim1_rigid_body. At first time, I used python3.8 and it turned out the same error. And I thus tried python3.10, and it made it! Thanks a lot for your kindness!

Cheers!
Lrc