stack-of-tasks/pinocchio #2171: Possible unnecessary computation in computeMinverse

alpylmz, 2024-03-15:
Hi, I am using Pinocchio to calculate inverse inertia matrix for my robot arm, and I am working on the computeMinverse function for research purposes. In `include/pinocchio/algorithm/aba.hxx` line 310, there is a computation that I couldn't exactly understand.
```
typedef typename SizeDepType<JointModel::NV>::template ColsReturn<typename Data::Matrix6x>::Type ColsBlock;
ColsBlock J_cols = jmodel.jointCols(data.J);
J_cols = data.oMi[i].act(jdata.S());
```
It seems it doesn't do anything useful, and if I delete this part my compuations for the Panda arm seems unaffected. Is there anyone who can verify that?

jcarpent, 2024-03-15:
What do you want to do exactly (could you provide full details)?

alpylmz, 2024-03-16:
It seems that this computation may only be necessary for certain types of robots. Perhaps we could encapsulate this computation within an if condition to check if it's required. If you think that could be useful, I can provide test cases for different robots to show it, and maybe someone who understands this can help us.

jcarpent, 2024-03-16:
My question was rather oriented towards your need: why do you need to compute the mass matrix inverse ?

alpylmz, 2024-03-17:
I have an optimal control based robot controller. I am using Minv in the constraints of the system. Currently I am only using a robot arm, but I'll use different robots in the future. Why do you ask?

jcarpent, 2024-03-17:
My guess is that you might not be obliged to compute this quantity.

closed 2024-03-18 by jcarpent