stack-of-tasks/pinocchio #2177: Generalized velocity of the floating base

thuuzi, 2024-03-19:
Hi,

After reviewing the discussion in https://github.com/stack-of-tasks/pinocchio/issues/1137, I understand that the velocity of the floating base is typically represented in the LOCAL frame. However, when I replaced a free flyer joint with a combination of JointModelTranslation() and JointModelSphericalZYX(), it appeared that the velocity is instead  expressed in the WORLD frame.

To clarify my confusion, I have outputted the Jacobian Matrix in the LOCAL_WORLD_ALIGNED frame, where the section corresponding to the floating base velocity consistently turns out to be a 3x3 identity matrix.Could you please confirm it?

Thank you in advance for your clarification.

jcarpent, 2024-03-19:
Could you provide a short Python example to understand what you did exactly?

fabinsch, 2024-03-19:
hi @thuuzi, after several discussions we have recently merged a [python script](https://github.com/stack-of-tasks/pinocchio/blob/080720adb012468ad4ff2956383199a6877608c7/examples/floating-base-velocity-viewer.py) in #2143 that could help you. You can easily extend it to have a combination of `JointModelTranslation` and `JointModelSphericalZYX` as root joint and visualize it.