stack-of-tasks/pinocchio #2181: pin.getJointJacobian

LIUHAITAO-CH, 2024-03-21:
import pinocchio as pin

from pinocchio.visualize import MeshcatVisualizer

urdf_path = "../urdf/Franka_urdf/fr3_without_hand.urdf"

model: pin.Model = pin.buildModelFromUrdf(urdf_path)

data = model.createData()

q0 = pin.neutral(model)

end_frame = model.getJointId("fr3_joint8")

pin.computeJointJacobians(model, data, q0)

end_jacobian = pin.getJointJacobian(model, data, end_frame, pin.LOCAL)

why did the following error occur? The IDE I used is pycharm and the system is macbook.

LIUHAITAO-CH, 2024-03-21:
[image]

fabinsch, 2024-03-21:
Hi @LIUHAITAO-CH, 
I think the problem might be coming from the variable `end_frame`. Could you check its value? and also check what value `len(model.joints)` has. If you use the `model.getJointId` and you provide a name of a joint that does not exist in the model, the function does not throw an error but the id you obtain is not correct and could like to an error later in your call to `getJointJacobian`.

LIUHAITAO-CH, 2024-03-22:
> Hi @LIUHAITAO-CH, I think the problem might be coming from the variable `end_frame`. Could you check its value? and also check what value `len(model.joints)` has. If you use the `model.getJointId` [...]

Hello @fabinsch ,
the picture  is my urdf file
[image]

I think what you said is right .the result of `end_frame = model.getJointId("fr3_joint8")` is 8;but the `fr3_joint8` is fixed joint.
I replaced `fr3_joint8` with  `fr3_joint7`,the result is ok.
thanks,and could you  tell me the why?

fabinsch, 2024-03-22:
In Pinocchio fixed joints are treated as frames. As stated in the [docs](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_a-features_intro.html).

> Remark: In the URDF format, a joint of type fixed can be defined. However, a fixed joint is not really a joint because it cannot move. For efficiency reasons, it is therefore treated as operational frame of the model.

In your case you should use the functions `getFrameId` and `getFrameJacobian` if you want the Jacobian at what is called `fr3_joint8` in your model.

closed 2024-03-22 by fabinsch

LIUHAITAO-CH, 2024-03-22:
ok, @fabinsch ,thanks your advices,i understand what you said,
but I have an new question: 

Did they all use `pin.computeJointJacobians(model, data, q)`  with model`.getJointId`、 `model.getFrameId` 、 `pin.getJointJacobian`and `pin.getFrameJacobian` ，could we use `pin.computeFrameJacobian(model, data, q)`

fabinsch, 2024-03-22:
Yes, you can choose the way that is better for your application. Please have a look at the docstrings of these functions: 
```
pin.getFrameJacobian?
Docstring:
getFrameJacobian( (Model)model, (Data)data, (int)frame_id, (ReferenceFrame)reference_frame) -> numpy.ndarray :
    Computes the Jacobian of the frame given by its ID either in the local or the world frames.
    The columns of the Jacobian are expressed in the LOCAL frame coordinates system.
    In other words, the velocity of the frame vF expressed in the local coordinate is given by J*v,where v is the joint velocity.
    computeJointJacobians(model,data,q) and updateFramePlacements(model,data) must have been called first.
Type:      function
```
You can use this if you need more than just the Jacobian of the end-effector. Remember to call `updateFramePlacements` as stated here.

If you only need the Jacobian of the end-effector, it might be better to use 
```
pin.computeFrameJacobian?
Docstring:
computeFrameJacobian( (Model)model, (Data)data, (numpy.ndarray)q, (int)frame_id, (ReferenceFrame)reference_frame) -> numpy.ndarray :
    Computes the Jacobian of the frame given by its frame_id in the coordinate system given by reference_frame.
    

computeFrameJacobian( (Model)model, (Data)data, (numpy.ndarray)q, (int)frame_id) -> numpy.ndarray :
    Computes the Jacobian of the frame given by its frame_id.
    The columns of the Jacobian are expressed in the coordinates system of the Frame itself.
    In other words, the velocity of the frame vF expressed in the local coordinate is given by J*v,where v is the joint velocity.
Type:      function
```

LIUHAITAO-CH, 2024-03-23:
thank you for your reply @fabinsch 

I think that i can get `getJointJacobian` `getFrameJacobian` by `computeJointJacobians` ,
`pin.computeJointJacobians(model, data, q_random)`
`Joint7_jacobian = pin.getJointJacobian(model, data, Joint7, pin.LOCAL)`
`end_jacobian = pin.getFrameJacobian(model, data, end_frame, pin.LOCAL)`

and i can directly get frame jacobian by  `computeFrameJacobian`,without `pin.getFrameJacobian`
end_jacobian_ = pin.computeFrameJacobian(model, data, q_random, end_frame, pin.LOCAL)

and i have a question 

1. `Joint7_jacobian = pin.getJointJacobian(model, data, Joint7, pin.LOCAL)`means the jacobian matrix of joint7 versus joint 0 (base)? pin.WORLD means what?

2. should `updateFramePlacements(model,datata）`appear with` computeJointJacobians(model,data,q)`？and During the control, they have to refresh every control cycle, right？

thank you very much

LIUHAITAO-CH, 2024-03-27:
> thank you for your reply @fabinsch
>
> I think that i can get `getJointJacobian` `getFrameJacobian` by `computeJointJacobians` , `pin.computeJointJacobians(model, data, q_random)` `Joint7_jacobian = [...]

@fabinsch hello,can you help me answer this question? thsnks

stephane-caron, 2024-03-27:
Hi @LIUHAITAO-CH,

I noticed that you have bumped this thread. While we appreciate your interest in this discussion, bumping threads puts pressure on fellow Pinocchio users who are answering on their free time.

Instead, consider asking only one question at a time, and when asking, show that you have done your research. For instance, when you ask:

> pin.WORLD means what?

Add some quotes from the parts of the [Pinocchio documentation](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/) where you have searched, and then only, ask for clarifications if anything is unclear. This approach is more constructive and will help you get better feedback.

In the same spirit, make sure you open [a separate discussion](https://github.com/stack-of-tasks/pinocchio/discussions) for each new single question.

LIUHAITAO-CH, 2024-03-27:
> Hi @LIUHAITAO-CH,
>
> I noticed that you have bumped this thread. While we appreciate your interest in this discussion, bumping threads puts pressure on fellow Pinocchio users who are answering on [...]

ok，thanks，i try