stack-of-tasks/pinocchio #2187: Fix cmake export to allow cross-compilation with sysroot

aurelien-enchanted-tools, 2024-03-25:
- Pinocchio version: from 2.7.0 and previous to the current.

## Bug description

In the case of cross-compilation with sysroot, cmake export generated files have bad paths to be imported into another project. This doesn't allow to link to it when you import it as dependence in another project. 

## Expected behavior

Cmake exports have to allow to cross-compilation with sysroot. 
(cf. for more description and an example https://github.com/humanoid-path-planner/hpp-fcl/issues/542#issuecomment-1979026071 ). Paths have to be relatives.

### Code

Please first fix the following
```diff
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index f2c30958..cdbeb5d4 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -97,12 +97,12 @@ ADD_HEADER_GROUP(${PROJECT_NAME}_CORE_GENERATED_PUBLIC_HEADERS)

 INSTALL(TARGETS ${PROJECT_NAME}
   EXPORT ${TARGETS_EXPORT_NAME}
-  LIBRARY DESTINATION ${CMAKE_INSTALL_FULL_LIBDIR}
-  ARCHIVE DESTINATION ${CMAKE_INSTALL_FULL_LIBDIR}
-  RUNTIME DESTINATION ${CMAKE_INSTALL_FULL_BINDIR})
+  LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR}
+  ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR}
+  RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})

 INSTALL(TARGETS ${PROJECT_NAME}_headers
   EXPORT ${TARGETS_EXPORT_NAME}
-  LIBRARY DESTINATION ${CMAKE_INSTALL_FULL_LIBDIR}
-  ARCHIVE DESTINATION ${CMAKE_INSTALL_FULL_LIBDIR}
-  RUNTIME DESTINATION ${CMAKE_INSTALL_FULL_BINDIR})
+  LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR}
+  ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR
+  RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})
```
But it is not enough, in your sources you have another place where lib path is forced as absolute. My current workaround is to modify manually the generate export files after build of pinocchio to allow to use it as dependence in another project, this files are:
- […]/pkgconfig/pinocchio.pc
- […]/cmake/pinocchio/pinocchioConfig.cmake
- […]/cmake/pinocchio/pinocchioTargets.cmake
Please fix also the cmake file which generates them.

Thank you in advance

jcarpent, 2024-03-25:
Thanks @aurelien-enchanted-tools for raising this issue.
@MegMll Could you handle this task on the devel branch?

jcarpent, 2024-04-24:
Solved via #2202.

closed 2024-04-24 by jcarpent