stack-of-tasks/pinocchio #2227: Release of Pinocchio 2.7.0 into ROS 2 Rolling (and possibly Jazzy)

clalancette, 2024-04-25:
Currently pinocchio is failing to build on ROS 2 Rolling: https://build.ros2.org/job/Rsrc_uN__pinocchio__ubuntu_noble__source/ .

The easiest solution to this problem is to do a new release of pinocchio into ROS 2 Rolling using `bloom-release`, which should bump the version to 2.7.0.

If you want, you can also do a release of pinocchio 2.7.0 into Jazzy; that is completely up to you.

Please let me know if you have any questions.

nim65s, 2024-04-25:
@clalancette thanks for stepping into this !

And sorry for not fixing this sooner.

Could you point me to a way to test a release before just running bloom ? I have been "just running bloom" a couple time on pinocchio and its dependencies (eigenpy & hpp-fcl), but this is not working better.

For ROS1 (we also have broken packages on noetic), I was using prerelease.ros.org, which was allowing me to fix the ROS patches before running bloom, but I can't anymore: https://github.com/ros-infrastructure/prerelease_website/issues/65

clalancette, 2024-04-26:
> Could you point me to a way to test a release before just running bloom to open a PR ? I have been "just running bloom" a couple time on pinocchio and its dependencies (eigenpy & hpp-fcl), but this is not working better.

Yeah, unfortunately there isn't currently an easy way.  In theory it is possible to do a pre-release type thing following the instructions in https://www.youtube.com/watch?v=XA20_iRBLrg , but I have no idea how well that works anymore.

I'll suggest just doing a bloom-release, and seeing what happens.  Totally understood that it isn't the most efficient way, but it is the best I can offer for now.

nim65s, 2024-04-26:
Thanks, I will follow your suggestion asap, and look at those instructions after that.

nim65s, 2024-09-14:
done: https://index.ros.org/p/pinocchio/#jazzy

closed 2024-09-14 by nim65s

wxmerkt, 2024-09-15:
@nim65s I saw the build failures reduce (a few on rolling remain) but 2.7 or newer not yet land in Jazzy or Rolling per the distribution files - 2.6.21 appears to be the most recent. The ROS2 release repo also doesn't have related PRs (https://github.com/ros/rosdistro/pulls?q=is%3Apr+pinocchio).

I ran the release for Rolling only against main (currently at 3.2), cf. https://github.com/ros/rosdistro/pull/42847 - please let me know if I should close this (please note no special flags set for Rolling, so it won't be build in v2 compatibility mode)