stack-of-tasks/pinocchio #2229: Make pinocchio python binding documentation online

hwyao, 2024-04-29:
## Python documentation 

While I am using this python bindings, I realize there are some [Cheatsheet](https://github.com/stack-of-tasks/pinocchio/blob/master/doc/pinocchio_cheat_sheet.pdf), which is quite helpful, but after running `dir` for each class, there are still some methods not recorded in this cheatsheet.

By searching on google, what we can get (only considering documentations form authors) are [Github pages](https://stack-of-tasks.github.io/pinocchio/), [pinocchio doxygen mainly for C++](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/index.html). [Pinocchio PyBind11 helpers](https://docs.ros.org/en/melodic/api/pinocchio/html/index.html). There are not too much extra information over the python interfaces apart from the cheatsheet)

Actually, everything about binding is already here in files of [include/pinocchio/bindings/python](https://github.com/stack-of-tasks/pinocchio/tree/master/include/pinocchio/bindings/python). (e.g. (e.g. [SE3.Interpolate](https://github.com/stack-of-tasks/pinocchio/blob/ac0b1aa6b18931bed60d0657de3c7680a4037dd3/include/pinocchio/bindings/python/spatial/se3.hpp#L165))). 

What I think better as a feature is just extract them out and make as a website and make it available somewhere, so that python classes and methods can be searched with google, and there are hyperlinks inside the websites, like the existing C++ documentation.

## Feature Request 

We can try to use tools like [pydoc](https://docs.python.org/3/library/pydoc.html) or [pdoc](https://pdoc3.github.io/pdoc/) to extract a website from it. All these commands are just based on the [robotpkg](https://stack-of-tasks.github.io/pinocchio/download.html) installation.

### pydoc
Like this (need some manual fixing). This is just some edit over the [mikecharles's script](https://gist.github.com/mikecharles/93b323bf045fbf43b157b917bab7987a).
```python
#!/usr/bin/env python
import pydoc
import os, sys
import pkgutil

def list_submodules(list_name, package_name):
    for loader, module_name, is_pkg in pkgutil.walk_packages(package_name.__path__, package_name.__name__+'.'):
        list_name.append(module_name)
        module_name = __import__(module_name, fromlist='dummylist')
        if is_pkg:
            list_submodules(list_name, module_name)

if len(sys.argv) != 2:
    print('Usage: {} [PACKAGE-NAME]'.format(os.path.basename(__file__)))
    sys.exit(1)
else:
    package_name = sys.argv[1]

try:
    package = __import__(package_name)
except ImportError:
    print('Package {} not found...'.format(package_name))
    sys.exit(1)

all_modules = [package_name]
list_submodules(all_modules, package)
for module in all_modules:
    print(module)
    pydoc.writedoc(module)

# Run this script with the following command:
# python3 generate_doc_pin.py pinocchio
# To fix some missing files run the following commands:
# python3 -m pydoc -w pinocchio.pinocchio_pywrap.cholesky pinocchio.pinocchio_pywrap.liegroups pinocchio.pinocchio_pywrap.rpy pinocchio.pinocchio_pywrap.serialization
```
Preview:
[image]
[image]
(I like the color and the link of the classes inside, but the process of the generation is a bit weird. `-w` parameter makes the program stuck so I have to do it recursively on my own).

[generated_pydoc.zip](https://github.com/stack-of-tasks/pinocchio/files/15147928/generated_pydoc.zip)

### pdoc
Or running this command: 
```bash
python3 -m pdoc --html pinocchio
# have many warnings, does have time to see why
```
Preview:
[image]
(I like the structured layout here but the table of contents is too long).

[generated_pdoc.zip](https://github.com/stack-of-tasks/pinocchio/files/15147939/generated_pdoc.zip)

### Into CI/CD process

Well…… it is still not the same quality as the doxygen files. But since the documentation here [Main features of Pinocchio/Python bindings](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_a-features_l-python.html) is still empty, maybe we can consider to make these automatic generated html somehow integrated or hyperlinked here for searching and clicking.

[image]

Hopefully we can discuss more things about this (and maybe play with some parameters and configurations to explore better documentation generation). :)

jcarpent, 2024-04-29:
Thanks for the report.
Here is a tentative: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/jnrh2023/template/algorithm.html

jcarpent, 2024-04-29:
We will come back to better solutions soon. Very sorry for the lack of clear Python docs.

hwyao, 2024-04-29:
Oh, you already have this here with Sphinx, that's great. 

Seems that the SEO of this site is a bit not optimized. Making this site better SEO maybe?

By trying some search with google：
```
pinocchio computeFrameJacobian
```
Would miss this site.

```
pinocchio computeFrameJacobian site:https://gepettoweb.laas.fr/
```
This would hit the site at the 3rd place.

```
pinocchio computeFrameJacobian site:https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/jnrh2023/
```
This would finally hit the only site available.

p.s. 
```
pinocchio computeFrameJacobian python site:https://gepettoweb.laas.fr/
```
Would miss this site.