stack-of-tasks/pinocchio #2231: Build error when installing with pip

WanqingXia, 2024-04-30:
I am using PyCharm virtual env to install pin, I met the following build error. I have tried to install cmeel but it does not fix the problem. Does anyone know how to fix this?

Thanks in advance.

(venvnew) (base) iai-lab@iailab-System-Product-Name:~/Documents/megapose6d$ pip install pin
'''
Collecting pin
  Using cached pin-2.7.0.tar.gz (61.8 MB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [62 lines of output]
      Collecting cmeel-boost~=1.83.0
        Using cached cmeel_boost-1.83.0.tar.gz (4.0 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Preparing metadata (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [39 lines of output]
            -- The C compiler identification is GNU 7.5.0
            -- The CXX compiler identification is GNU 7.5.0
            -- Detecting C compiler ABI info
            -- Detecting C compiler ABI info - done
            -- Check for working C compiler: /usr/bin/cc - skipped
            -- Detecting C compile features
            -- Detecting C compile features - done
            -- Detecting CXX compiler ABI info
            -- Detecting CXX compiler ABI info - done
            -- Check for working CXX compiler: /usr/bin/c++ - skipped
            -- Detecting CXX compile features
            -- Detecting CXX compile features - done
            CMake Error at /tmp/pip-build-env-s9j092i6/overlay/lib/python3.9/site-packages/cmake/data/share/cmake-3.29/Modules/FindPython/Support.cmake:719 (file):
              file STRINGS file
              "/home/iai-lab/Downloads/Python-3.9.19/Include/pyconfig.h" cannot be read.
            Call Stack (most recent call first):
              /tmp/pip-build-env-s9j092i6/overlay/lib/python3.9/site-packages/cmake/data/share/cmake-3.29/Modules/FindPython/Support.cmake:3569 (_python_get_version)
              /tmp/pip-build-env-s9j092i6/overlay/lib/python3.9/site-packages/cmake/data/share/cmake-3.29/Modules/FindPython3.cmake:545 (include)
              CMakeLists.txt:11 (find_package)
      
      
            -- Found Python3: /home/iai-lab/Documents/SiameseViT/venvnew/bin/python (found version "3.9.19") found components: Interpreter Development.Module
            -- CONFIGURE_COMMAND: ./bootstrap.sh;--prefix=/tmp/cmeel-6rb9gsy2/whl/cmeel.prefix
            -- BUILD_COMMAND: ./b2;hardcode-dll-paths=true;dll-path='$ORIGIN';link=shared;python=3.9
            -- Configuring incomplete, errors occurred!
            Traceback (most recent call last):
              File "/home/iai-lab/Documents/SiameseViT/venvnew/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
                main()
              File "/home/iai-lab/Documents/SiameseViT/venvnew/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
                json_out['return_val'] = hook(**hook_input['kwargs'])
              File "/home/iai-lab/Documents/SiameseViT/venvnew/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
                whl_basename = backend.build_wheel(metadata_directory, config_settings)
              File "/tmp/pip-build-env-s9j092i6/overlay/lib/python3.9/site-packages/cmeel/build.py", line 26, in build_wheel
                return build_impl(wheel_directory, editable=False)
              File "/tmp/pip-build-env-s9j092i6/overlay/lib/python3.9/site-packages/cmeel/impl.py", line 100, in build_impl
                check_call(configure_cmd, env=configure_env)
              File "/home/iai-lab/Downloads/Python-3.9.19/Lib/subprocess.py", line 373, in check_call
                raise CalledProcessError(retcode, cmd)
            subprocess.CalledProcessError: Command '['cmake', '-S', '.', '-B', '/tmp/cmeel-6rb9gsy2/bld', '-DBoost_NO_WARN_NEW_VERSIONS=ON', '-DCMAKE_BUILD_TYPE=Release', '-DCMAKE_INSTALL_LIBDIR=lib', '-DCMAKE_INSTALL_PREFIX=/tmp/cmeel-6rb9gsy2/whl/cmeel.prefix', '-DPYTHON_SITELIB=lib/python3.9/site-packages', '-DPython3_EXECUTABLE=/home/iai-lab/Documents/SiameseViT/venvnew/bin/python', '-DCMAKE_APPLE_SILICON_PROCESSOR=arm64', '-DCMEEL_JOBS=4', '-DBUILD_TESTING=OFF']' returned non-zero exit status 1.
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: metadata-generation-failed
      
      × Encountered error while generating package metadata.
      ╰─> See above for output.
      
      note: This is an issue with the package mentioned above, not pip.
      hint: See above for details.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
'''

### System
- OS: Ubuntu 20.04
- Python version: 3.9

jcarpent, 2024-04-30:
One quick alternative is to rely on conda and use `conda install -c conda-forge pinocchio`.

joao-pm-santos96, 2024-05-09:
Hi @jcarpent ! If I want to help fix this issue, where should I start?

joao-pm-santos96, 2024-05-09:
My apologies, I missed understood the original issue. I didn't realize that it was a local installation and assumed that @WanqingXia was talking about some GitHub Action that should publish Pinocchio to pypi but was falling. Short answer, I do not have this issue. Once again, my apologies...

nim65s, 2024-05-09:
> you can then try to `pip install cmeel-boost` directly

This is not the right time: I am currently upgrading cmeel-boost. It will take some time while cmeel boost 1.84 is available, and eigenpy and pinocchio binaries are not ready for boost 1.84.

@WanqingXia : we have working binaries, so I'd first like to know why you are building from source. Can you provide your python version, pip version, OS and architecture ?

nim65s, 2024-05-09:
Ok, I saw 20.04 (I assume x86_64) and py 3.9. How did you install python 3.9 on ubuntu 20.04 ?

WanqingXia, 2024-05-09:
> Ok, I saw 20.04 (I assume x86_64) and py 3.9. How did you install python 3.9 on ubuntu 20.04 ?

Thank you for still tracking this issue. I downloaded python 3.9 from the official website [Gzipped source tarball](https://www.python.org/ftp/python/3.9.16/Python-3.9.16.tgz). After installing it, I used it as the interpreter in my PyCharm venv.

nim65s, 2024-05-10:
Ok, I tried to reproduce your issue in a Dockerfile, and it is working for me. Is there anything you did differently ?

```Dockerfile
FROM ubuntu:20.04

ADD https://www.python.org/ftp/python/3.9.16/Python-3.9.16.tgz /

# https://devguide.python.org/getting-started/setup-building/#build-dependencies
RUN echo 'deb-src http://archive.ubuntu.com/ubuntu/ focal main' >> /etc/apt/sources.list
RUN --mount=type=cache,sharing=locked,target=/var/cache/apt \
    --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -y && DEBIAN_FRONTEND=noninteractive apt-get install -qqy --no-install-recommends \
    build-essential gdb lcov pkg-config \
    libbz2-dev libffi-dev libgdbm-dev libgdbm-compat-dev liblzma-dev \
    libncurses5-dev libreadline6-dev libsqlite3-dev libssl-dev \
    lzma lzma-dev tk-dev uuid-dev zlib1g-dev libmpdec-dev

RUN tar xf /Python-3.9.16.tgz
WORKDIR /Python-3.9.16
RUN ./configure \
 && make \
 && make install

RUN python3 -m venv venv \
 && . venv/bin/activate \
 && pip install pin

WanqingXia, 2024-05-13:
[image]
I tried to create a new venv with python 3.9 and install pip, the problem still exists. Here is the full output

```
(venv) iai-lab@iailab-System-Product-Name:~/Downloads$ pip install pin
Collecting pin
  Using cached pin-2.7.0.tar.gz (61.8 MB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [160 lines of output]
      Collecting cmeel-boost~=1.83.0
        Using cached cmeel_boost-1.83.0-0-cp39-cp39-manylinux_2_27_x86_64.whl
      Collecting cmeel-urdfdom[build]<4,>=3.1.1.1
        Using cached cmeel_urdfdom-3.1.1.1.tar.gz (292 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting cmeel[build]
        Using cached cmeel-0.53.3-py3-none-any.whl (20 kB)
      Collecting hpp-fcl[build]<4,>=2.3.4
        Using cached hpp_fcl-2.4.4.tar.gz (3.2 MB)
        Installing build dependencies: started
        Installing build dependencies: still running...
        Installing build dependencies: still running...
        Installing build dependencies: finished with status 'error'
        error: subprocess-exited-with-error
      
        × pip subprocess to install build dependencies did not run successfully.
        │ exit code: 1
        ╰─> [124 lines of output]
            Collecting cmeel-assimp<6,>=5.3.1
              Using cached cmeel_assimp-5.3.1.tar.gz (52.6 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
              Getting requirements to build wheel: started
              Getting requirements to build wheel: finished with status 'done'
              Preparing metadata (pyproject.toml): started
              Preparing metadata (pyproject.toml): still running...
              Preparing metadata (pyproject.toml): still running...
              Preparing metadata (pyproject.toml): finished with status 'done'
            Collecting cmeel-boost~=1.83.0
              Using cached cmeel_boost-1.83.0-0-cp39-cp39-manylinux_2_27_x86_64.whl
            Collecting cmeel-octomap<2,>=1.9.8.2
              Using cached cmeel_octomap-1.9.8.2.tar.gz (1.7 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
              Getting requirements to build wheel: started
              Getting requirements to build wheel: finished with status 'done'
              Preparing metadata (pyproject.toml): started
              Preparing metadata (pyproject.toml): finished with status 'done'
            Collecting cmeel-qhull<8.0.3,>=8.0.2.1
              Using cached cmeel_qhull-8.0.2.1.tar.gz (1.3 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
              Getting requirements to build wheel: started
              Getting requirements to build wheel: finished with status 'done'
              Preparing metadata (pyproject.toml): started
              Preparing metadata (pyproject.toml): finished with status 'done'
            Collecting cmeel[build]
              Using cached cmeel-0.53.3-py3-none-any.whl (20 kB)
            Collecting eigenpy[build]<4,>=3.1
              Using cached eigenpy-3.5.1.tar.gz (1.0 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
[... 92 lines cut ...]
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: subprocess-exited-with-error
      
      × pip subprocess to install build dependencies did not run successfully.
      │ exit code: 1
      ╰─> See above for output.
      
      note: This error originates from a subprocess, and is likely not a problem with pip.
      
      [notice] A new release of pip is available: 23.0.1 -> 24.0
      [notice] To update, run: pip install --upgrade pip
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip is available: 23.0.1 -> 24.0
[notice] To update, run: pip install --upgrade pip
```

WanqingXia, 2024-05-13:
I also tried to manually install cmeel-boost first before installing pin
[image]
But the output suggest the same error
```

(venv) iai-lab@iailab-System-Product-Name:~/Downloads$ pip install pin
Collecting pin
  Using cached pin-2.7.0.tar.gz (61.8 MB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [75 lines of output]
      Collecting cmeel-boost~=1.83.0
        Using cached cmeel_boost-1.83.0-0-cp39-cp39-manylinux_2_27_x86_64.whl
      Collecting cmeel-urdfdom[build]<4,>=3.1.1.1
        Using cached cmeel_urdfdom-3.1.1.1.tar.gz (292 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Preparing metadata (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [47 lines of output]
            -- The CXX compiler identification is GNU 7.5.0
            -- The C compiler identification is GNU 7.5.0
            -- Detecting CXX compiler ABI info
            -- Detecting CXX compiler ABI info - done
            -- Check for working CXX compiler: /usr/bin/c++ - skipped
            -- Detecting CXX compile features
            -- Detecting CXX compile features - done
            -- Detecting C compiler ABI info
            -- Detecting C compiler ABI info - done
            -- Check for working C compiler: /usr/bin/cc - skipped
            -- Detecting C compile features
            -- Detecting C compile features - done
            -- urdfdom version 3.0.2
            -- Performing Test HAS_VISIBILITY
            -- Performing Test HAS_VISIBILITY - Success
            CMake Error at CMakeLists.txt:48 (find_package):
              By not providing "Findtinyxml.cmake" in CMAKE_MODULE_PATH this project has
              asked CMake to find a package configuration file provided by "tinyxml", but
              CMake did not find one.
      
              Could not find a package configuration file provided by "tinyxml" with any
              of the following names:
      
                tinyxmlConfig.cmake
                tinyxml-config.cmake
      
              Add the installation prefix of "tinyxml" to CMAKE_PREFIX_PATH or set
              "tinyxml_DIR" to a directory containing one of the above files.  If
              "tinyxml" provides a separate development package or SDK, be sure it has
              been installed.
      
      
            -- Configuring incomplete, errors occurred!
            Traceback (most recent call last):
              File "/home/iai-lab/Downloads/venv/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
                main()
              File "/home/iai-lab/Downloads/venv/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
                json_out['return_val'] = hook(**hook_input['kwargs'])
              File "/home/iai-lab/Downloads/venv/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
                whl_basename = backend.build_wheel(metadata_directory, config_settings)
              File "/tmp/pip-build-env-xjo43wfa/overlay/lib/python3.9/site-packages/cmeel/build.py", line 26, in build_wheel
                return build_impl(wheel_directory, editable=False)
              File "/tmp/pip-build-env-xjo43wfa/overlay/lib/python3.9/site-packages/cmeel/impl.py", line 100, in build_impl
                check_call(configure_cmd, env=configure_env)
              File "/usr/local/lib/python3.9/subprocess.py", line 373, in check_call
                raise CalledProcessError(retcode, cmd)
            subprocess.CalledProcessError: Command '['cmake', '-S', 'urdfdom', '-B', '/tmp/cmeel-e_jmovi6/bld', '-DBoost_NO_WARN_NEW_VERSIONS=ON', '-DCMAKE_BUILD_TYPE=Release', '-DCMAKE_INSTALL_LIBDIR=lib', '-DCMAKE_INSTALL_PREFIX=/tmp/cmeel-e_jmovi6/whl/cmeel.prefix', '-DPYTHON_SITELIB=lib/python3.9/site-packages', '-DPython3_EXECUTABLE=/home/iai-lab/Downloads/venv/bin/python3.9', '-DCMAKE_APPLE_SILICON_PROCESSOR=arm64', '-DCMEEL_JOBS=4']' returned non-zero exit status 1.
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: metadata-generation-failed
      
      × Encountered error while generating package metadata.
      ╰─> See above for output.
      
      note: This is an issue with the package mentioned above, not pip.
      hint: See above for details.
      
      [notice] A new release of pip is available: 23.0.1 -> 24.0
      [notice] To update, run: pip install --upgrade pip
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip is available: 23.0.1 -> 24.0
[notice] To update, run: pip install --upgrade pip

```

nim65s, 2024-05-13:
- You use a `python3.9 activate venv` command, which does not exist, so this is certainly not doing what you expect: https://docs.python.org/3/library/venv.html#how-venvs-work
- You say you "install pip", but your logs plainly display your pip is not up to date: `A new release of pip is available: 23.0.1 -> 24.0`, so don't know what you actually did, but that was wrong. And it is not necessary.

Please just try again correctly. There are 3 commands to run, and I already provided those:

```
python3.9 -m venv new-venv
source new-venv/bin/activate
pip install pin

WanqingXia, 2024-05-14:
iai-lab@iailab-System-Product-Name:~/Downloads$ python3.9 -m venv new-venv
iai-lab@iailab-System-Product-Name:~/Downloads$ source new-venv/bin/activate
(new-venv) iai-lab@iailab-System-Product-Name:~/Downloads$ pip install pin

```
Collecting pin
  Using cached pin-2.7.0.tar.gz (61.8 MB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [160 lines of output]
      Collecting cmeel-boost~=1.83.0
        Using cached cmeel_boost-1.83.0-0-cp39-cp39-manylinux_2_27_x86_64.whl
      Collecting cmeel-urdfdom[build]<4,>=3.1.1.1
        Using cached cmeel_urdfdom-3.1.1.1.tar.gz (292 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting cmeel[build]
        Using cached cmeel-0.53.3-py3-none-any.whl (20 kB)
      Collecting hpp-fcl[build]<4,>=2.3.4
        Using cached hpp_fcl-2.4.4.tar.gz (3.2 MB)
        Installing build dependencies: started
        Installing build dependencies: still running...
        Installing build dependencies: still running...
        Installing build dependencies: finished with status 'error'
        error: subprocess-exited-with-error
      
        × pip subprocess to install build dependencies did not run successfully.
        │ exit code: 1
        ╰─> [124 lines of output]
            Collecting cmeel-assimp<6,>=5.3.1
              Using cached cmeel_assimp-5.3.1.tar.gz (52.6 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
              Getting requirements to build wheel: started
              Getting requirements to build wheel: finished with status 'done'
              Preparing metadata (pyproject.toml): started
              Preparing metadata (pyproject.toml): still running...
              Preparing metadata (pyproject.toml): still running...
              Preparing metadata (pyproject.toml): finished with status 'done'
            Collecting cmeel-boost~=1.83.0
              Using cached cmeel_boost-1.83.0-0-cp39-cp39-manylinux_2_27_x86_64.whl
            Collecting cmeel-octomap<2,>=1.9.8.2
              Using cached cmeel_octomap-1.9.8.2.tar.gz (1.7 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
              Getting requirements to build wheel: started
              Getting requirements to build wheel: finished with status 'done'
              Preparing metadata (pyproject.toml): started
              Preparing metadata (pyproject.toml): finished with status 'done'
            Collecting cmeel-qhull<8.0.3,>=8.0.2.1
              Using cached cmeel_qhull-8.0.2.1.tar.gz (1.3 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
              Getting requirements to build wheel: started
              Getting requirements to build wheel: finished with status 'done'
              Preparing metadata (pyproject.toml): started
              Preparing metadata (pyproject.toml): finished with status 'done'
            Collecting cmeel[build]
              Using cached cmeel-0.53.3-py3-none-any.whl (20 kB)
            Collecting eigenpy[build]<4,>=3.1
              Using cached eigenpy-3.5.1.tar.gz (1.0 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
[... 93 lines cut ...]
      error: subprocess-exited-with-error
      
      × pip subprocess to install build dependencies did not run successfully.
      │ exit code: 1
      ╰─> See above for output.
      
      note: This error originates from a subprocess, and is likely not a problem with pip.
      
      [notice] A new release of pip is available: 23.0.1 -> 24.0
      [notice] To update, run: pip install --upgrade pip
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip is available: 23.0.1 -> 24.0
[notice] To update, run: pip install --upgrade pip
```
(new-venv) iai-lab@iailab-System-Product-Name:~/Downloads$

nim65s, 2024-05-14:
Thanks. Can I don't understand why it can download some binaries but not others.

Please show me the output of this in the venv:
```
pip install packaging
python -c "from packaging.tags import sys_tags; print(next(sys_tags()))"

WanqingXia, 2024-05-14:
Here is the output
```
iai-lab@iailab-System-Product-Name:~/Downloads$ source new-venv/bin/activate
(new-venv) iai-lab@iailab-System-Product-Name:~/Downloads$ pip install packagingCollecting packaging
  Using cached packaging-24.0-py3-none-any.whl (53 kB)
Installing collected packages: packaging
Successfully installed packaging-24.0

[notice] A new release of pip is available: 23.0.1 -> 24.0
[notice] To update, run: pip install --upgrade pip
(new-venv) iai-lab@iailab-System-Product-Name:~/Downloads$ python -c "from packaging.tags import sys_tags; print(next(sys_tags()))"
cp39-cp39-manylinux_2_27_x86_64
```

stephane-caron, 2024-05-23:
Chiming in that we encounter the same error when trying to install Pinocchio from PyPI in ``macos-latest`` GitHub runner images ([runner log](https://github.com/stephane-caron/pink/actions/runs/9141838768/job/25323489540)): 
[raw_log.txt](https://github.com/stack-of-tasks/pinocchio/files/15415337/raw_log.txt)

I guess the relevant part is:

```
                     -- Checking for SciPy
                    CMake Error at cmake/python.cmake:544 (message):
                      Failed to detect scipy
```

emrecemal, 2024-06-20:
Hello

I also started to get this error today, before I had no issue. I'm using Macbook M1, with python 3.8. I see some commits about disabling macOS, I hope it's not related to that. Here is the output:

```
(.venv) ➜  project git:(master) ✗ pip install pin
Collecting pin
  Using cached pin-2.7.0.tar.gz (61.8 MB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [82 lines of output]
      Collecting cmeel-boost~=1.83.0
        Using cached cmeel_boost-1.83.0-0-cp38-cp38-macosx_11_0_arm64.whl.metadata (1000 bytes)
      Collecting cmeel-urdfdom<4,>=3.1.1.1 (from cmeel-urdfdom[build]<4,>=3.1.1.1)
        Using cached cmeel_urdfdom-3.1.1.1-0-py3-none-macosx_12_0_arm64.whl.metadata (2.8 kB)
      Collecting hpp-fcl<4,>=2.3.4 (from hpp-fcl[build]<4,>=2.3.4)
        Using cached hpp_fcl-2.4.4.tar.gz (3.2 MB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'error'
        error: subprocess-exited-with-error
      
        × pip subprocess to install build dependencies did not run successfully.
        │ exit code: 1
        ╰─> [59 lines of output]
            Collecting cmeel-assimp<6,>=5.3.1
              Using cached cmeel_assimp-5.3.1-0-py3-none-macosx_12_0_arm64.whl.metadata (884 bytes)
            Collecting cmeel-boost~=1.83.0
              Using cached cmeel_boost-1.83.0-0-cp38-cp38-macosx_11_0_arm64.whl.metadata (1000 bytes)
            Collecting cmeel-octomap<2,>=1.9.8.2
              Using cached cmeel_octomap-1.9.8.2-3-py3-none-macosx_12_0_arm64.whl.metadata (926 bytes)
            Collecting cmeel-qhull<8.0.3,>=8.0.2.1
              Using cached cmeel_qhull-8.0.2.1-0-py3-none-macosx_12_0_arm64.whl.metadata (838 bytes)
            Collecting eigenpy<4,>=3.1 (from eigenpy[build]<4,>=3.1)
              Using cached eigenpy-3.5.1.tar.gz (1.0 MB)
              Installing build dependencies: started
              Installing build dependencies: finished with status 'done'
              Getting requirements to build wheel: started
              Getting requirements to build wheel: finished with status 'done'
              Preparing metadata (pyproject.toml): started
              Preparing metadata (pyproject.toml): finished with status 'error'
              error: subprocess-exited-with-error
      
              × Preparing metadata (pyproject.toml) did not run successfully.
              │ exit code: 1
              ╰─> [28 lines of output]
                  -- JRL cmakemodules found in 'cmake/' git submodule
                  -- Configuring "eigenpy" (http://github.com/stack-of-tasks/eigenpy)
                  -- Package version (ROS package.xml): 3.5.1
                  CMake Error at /private/var/folders/yc/8l7611vj53g7dqtkh2m1807r0000gp/T/pip-build-env-u4wcenbz/overlay/lib/python3.8/site-packages/cmake/data/share/cmake-3.29/Modules/CMakeDetermineCXXCompiler.cmake:48 (message):
                    Could not find compiler set in environment variable CXX:
      
                    /usr/local/opt/llvm/bin/clang++.
      
                  Call Stack (most recent call first):
                    CMakeLists.txt:69 (project)
      
      
                  CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
                  -- Configuring incomplete, errors occurred!
                  Traceback (most recent call last):
                    File "/Users/emrecemal/project/.venv/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
                      main()
                    File "/Users/emrecemal/project/.venv/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
                      json_out['return_val'] = hook(**hook_input['kwargs'])
                    File "/Users/emrecemal/project/.venv/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
                      whl_basename = backend.build_wheel(metadata_directory, config_settings)
                    File "/private/var/folders/yc/8l7611vj53g7dqtkh2m1807r0000gp/T/pip-build-env-u4wcenbz/overlay/lib/python3.8/site-packages/cmeel/build.py", line 26, in build_wheel
[... 12 lines cut ...]
            ╰─> See above for output.
      
            note: This is an issue with the package mentioned above, not pip.
            hint: See above for details.
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: subprocess-exited-with-error
      
      × pip subprocess to install build dependencies did not run successfully.
      │ exit code: 1
      ╰─> See above for output.
      
      note: This error originates from a subprocess, and is likely not a problem with pip.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

emrecemal, 2024-06-20:
> Hello
>
> I also started to get this error today, before I had no issue. I'm using Macbook M1, with python 3.8. I see some commits about disabling macOS, I hope it's not related to that. Here is the [...]

UPDATE: I changed the python version from 3.8 to 3.12. Now, it's installed!

cyoahs, 2024-07-19:
Upgrading pip works for me.