stack-of-tasks/pinocchio #2234: Memory alignment issue when instantiating Data with AVX2

duburcqa, 2024-05-13:
## Bug description

Python is crashing when instantiating a Data from Python when pinocchio has been compiled in Debug with AVX2 instruction set enabled. 

It is only occurring at Python-side. I have no issue when running a C++ executable that has been compiled with the exact same flags. It also works fine when calling `pin.Model().createData()` in place of `pin.Data()`. So it means to be related to some `eigenpy` memory alignment issue. It is not systematically crashing the first time I try to create Data, sometimes it takes 2 or 3 trials. 

I already observed this issue I will ago, and I just decided to ignore it at that time because it was not blocking me. It is still not blocking me but it would be nice if it could be fixed, mainly because I don't understandard what is happening, and I don't like it.
 
## Reproduction steps

First, compile eigenpy and pinocchio with flags `BUILD_TYPE="Debug"` and `CMAKE_CXX_FLAGS="-march=x86-64-v3 -mtune=generic"`.

### Code

```python
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.10.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pinocchio as pin

In [2]: pin.Data(pin.Model())
Out[2]: <pinocchio.pinocchio_pywrap.Data at 0x7f2370a0f150>

In [3]: pin.Data(pin.Model())
python3: /home/duburcqa/workspace/src/jiminy_2/install/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; in
t Size = 36; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily
.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Aborted (core dumped)
```

### System

- OS: Ubuntu 22.04, Python 3.10.12
- Pinocchio version:  pinocchio 2.7.0, eigenpy 3.4.0

jcarpent, 2024-05-13:
It should be fixed in Pinocchio 3, coming soon.

duburcqa, 2024-05-13:
Any chance this could be backported ? I could take care of it eventually if I could understand the issue.

ManifoldFR, 2024-05-13:
What is your version of Boost?

duburcqa, 2024-05-13:
boost-1.76.0

ManifoldFR, 2024-05-13:
Hmm, I'm not sure whether that version works. From what I recall this is Boost.Python + Eigen issue that got patched by Justin a year or so ago in eigenpy.

Last I checked 1.74 and 1.75 still crap out with AVX2 but 1.78 and above work for all versions of eigenpy post-fix.

duburcqa, 2024-05-13:
OK I can confirm this. The precompiled version of pinocchio available on robotpkg it is using some boost version older than 1.78.0 as far as I know. Would it be possible to update it in order to maintain consistency with what is considered supported now ?

ManifoldFR, 2024-05-13:
I don't know about robotpkg, sorry :/

duburcqa, 2024-05-14:
I can confirm that using boost 1.78 is fixing the issue.

duburcqa, 2024-05-14:
Currently I am bit stuck on 1.76 because I want to maintain compatibility with robotpkg though... but at least it is good to know that the issue will vanish at some point.

ManifoldFR, 2024-05-14:
Yes, this has also been a huge problem for us when using Ros noetic unfortunately...

duburcqa, 2024-05-14:
For the record, I can confirm the fix was introduced with boost 1.78.0. [here](https://github.com/boostorg/python/pull/360) is the corresponding PR.

duburcqa, 2024-05-14:
I guess we can close this issue since nothing can be done about it except waiting for boost<1.78 to disappear.

closed 2024-05-14 by duburcqa

nim65s, 2024-05-14:
@duburcqa : robotpkg just use the version from your OS, so upgrading your OS will upgrade boost

duburcqa, 2024-05-14:
Yes, I figured this out when trying to understand from where it was getting the boost version. That is why I mentioned that it will be fine for Ubuntu 24.04, but we are out of luck for older Ubuntu releases...