stack-of-tasks/pinocchio #2250: Building on OSX fails

budzianowski, 2024-05-29:
## Bug description
Building from source on python 3.12 gives error:
pinocchio/src/parsers/mjcf/mjcf-graph.cpp:41:42: error: call to deleted member function 'get_child'

## Reproduction steps

Steps to reproduce the behavior:
Running according to the instructions:
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local
make -j$(nproc)

### System

- OS: Mac3
- Pinocchio version: 3.0.0

jorisv, 2024-05-29:
Hello,

Can you give me :
- all the dependencies version you are using
- your compiler and compiler version
- how do you have installed them (brew, conda, pip)
- what is Mac3 ?
- what CPU architecture is used (ARM, X64 ?)

Pinocchio 3 is building well on the macos CI agent that run macOS 14 Arm64.

budzianowski, 2024-05-29:
Hi @jorisv  - yes, I saw the building and that what puzzles me. I can build if I delete all mjcf parser source code.
My setup is:
cmake version 3.29.3
all dependencies (eigen boost) installed from brew
mac3 pro chip
arm

Do you mean python dependencies?

budzianowski, 2024-05-29:
Building without mjcf ends up with architecture complaints:

> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.12/site-packages/pinocchio/__init__.py", line 18, in <module>
    from .pinocchio_pywrap_default import *
ImportError: dlopen(/usr/local/lib/python3.12/site-packages/pinocchio/pinocchio_pywrap_default.cpython-312-darwin.so, 0x0002): tried: '/usr/local/lib/python3.12/site-packages/pinocchio/pinocchio_pywrap_default.cpython-312-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/python3.12/site-packages/pinocchio/pinocchio_pywrap_default.cpython-312-darwin.so' (no such file), '/usr/local/lib/python3.12/site-packages/pinocchio/pinocchio_pywrap_default.cpython-312-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/usr/local/lib/python3.12/site-packages/pinocchio/pinocchio_pywrap_default.3.0.0.cpython-312-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/python3.12/site-packages/pinocchio/pinocchio_pywrap_default.3.0.0.cpython-312-darwin.so' (no such file), '/usr/local/lib/python3.12/site-packages/pinocchio/pinocchio_pywrap_default.3.0.0.cpython-312-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))

Do you have some documentation how to build 3.0.0 cause I'm trying to follow: https://stack-of-tasks.github.io/pinocchio/download.html

jorisv, 2024-05-29:
@budzianowski Can you try to build from this [PR](https://github.com/stack-of-tasks/pinocchio/pull/2255) ?

budzianowski, 2024-05-29:
The PR solves the issue.

budzianowski, 2024-05-29:
I think the documentation (https://stack-of-tasks.github.io/pinocchio/download.html) should be updated to the latest version of the workflow:

> conda env create -f environment.yaml
> cmake .. \
>   -DCMAKE_CXX_COMPILER_LAUNCHER=ccache \
>   -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX \
>   -DBUILD_WITH_COLLISION_SUPPORT=ON \
>   -DBUILD_WITH_CASADI_SUPPORT=ON \
>   -DBUILD_WITH_AUTODIFF_SUPPORT=ON
> cmake --build . -j2
> ctest --output-on-failure
> cmake --install .
>

jorisv, 2024-05-29:
Right now the documentation is kind of fragmented and out dated, we are aware of it and we will work on it soon.

closed 2024-05-29 by budzianowski

nim65s, 2024-05-30:
> I believe the docs are handled in the separate repo?

No, they are in this repo, in the [`gh-pages`](https://github.com/stack-of-tasks/pinocchio/tree/gh-pages) branch