stack-of-tasks/pinocchio #2270: Installation of pinocchio 3.0

paumarquez, 2024-06-03:
## Bug description
Installing pinocchio 3.0 is not possible from pypi, conda or apt. When I do so, I get the 2.7 version

## Expected behavior
Pinocchio 3.0 should be installed.

## Reproduction steps

```
apt install -qqy lsb-release curl \
     && mkdir -p /etc/apt/keyrings \
     && curl http://robotpkg.openrobots.org/packages/debian/robotpkg.asc \
      | tee /etc/apt/keyrings/robotpkg.asc \
     && echo "deb [arch=amd64 signed-by=/etc/apt/keyrings/robotpkg.asc] http://robotpkg.openrobots.org/packages/debian/pub $(lsb_release -cs) robotpkg" \
      | tee /etc/apt/sources.list.d/robotpkg.list \
     && apt update \
     && apt install -qqy robotpkg-py3*-pinocchio

export PATH=/opt/openrobots/bin:$PATH
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/opt/openrobots/lib/python3.10/site-packages:$PYTHONPATH # Adapt your desired python version here
export CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH
```
In python:
```
import pinocchio as pin                                                                                                                  
pin.__version__                                                                                                                          
'2.7.0'
```

### System

- Ubuntu 22.04

nim65s, 2024-06-03:
This is known, we are working on it.
ref https://github.com/stack-of-tasks/pinocchio/discussions/2265

jcarpent, 2024-06-27:
@paumarquez You can now use conda to get Pinocchio 3 on Mac, Linux, and Windows systems. Enjoy!
I will close this issue. Meanwhile, alternative installation solutions will come with time.

closed 2024-06-27 by jcarpent

joao-pm-santos96, 2024-07-11:
Hi there!
Is there any way that I can help to make pinocchio 3.* available though pipy?
Thank you!

matheecs, 2024-07-11:
> Hi there! Is there any way that I can help to make pinocchio 3.* available though pipy? Thank you!

please refer to https://github.com/stack-of-tasks/pinocchio/discussions/2265