stack-of-tasks/pinocchio #2271: pinocchio 3.0.0 build failure

chenrui333, 2024-06-03:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>error build log</summary>

```
  [ 34%] Building CXX object src/CMakeFiles/pinocchio_parsers.dir/parsers/urdf/geometry.cpp.o
  cd /tmp/pinocchio-20240527-10090-tqhq72/pinocchio-3.0.0/build/src && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DHPP_FCL_HAS_OCTOMAP -DHPP_FCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=10 -DOCTOMAP_PATCH_VERSION=0 -DPINOCCHIO_CONTEXT_FILE=\"pinocchio/context/default.hpp\" -DPINOCCHIO_ENABLE_TEMPLATE_INSTANTIATION -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -Dpinocchio_parsers_EXPORTS -I/tmp/pinocchio-20240527-10090-tqhq72/pinocchio-3.0.0/build -I/tmp/pinocchio-20240527-10090-tqhq72/pinocchio-3.0.0/build/include -I/tmp/pinocchio-20240527-10090-tqhq72/pinocchio-3.0.0/include -isystem /opt/homebrew/include/eigen3 -isystem /opt/homebrew/include/urdfdom -isystem /opt/homebrew/include/urdfdom_headers -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -std=gnu++14 -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -fPIC -MD -MT src/CMakeFiles/pinocchio_parsers.dir/parsers/urdf/geometry.cpp.o -MF CMakeFiles/pinocchio_parsers.dir/parsers/urdf/geometry.cpp.o.d -o CMakeFiles/pinocchio_parsers.dir/parsers/urdf/geometry.cpp.o -c /tmp/pinocchio-20240527-10090-tqhq72/pinocchio-3.0.0/src/parsers/urdf/geometry.cpp
  /tmp/pinocchio-20240527-10090-tqhq72/pinocchio-3.0.0/src/parsers/mjcf/mjcf-graph.cpp:40:40: error: call to deleted member function 'get_child'
              ptree attr_parent = parent.get_child(path, ptree());
                                  ~~~~~~~^~~~~~~~~
  /opt/homebrew/include/boost/property_tree/ptree.hpp:274:14: note: candidate function has been explicitly deleted
          void get_child(const path_type &path,
               ^
  /opt/homebrew/include/boost/property_tree/ptree.hpp:270:26: note: candidate function
          const self_type &get_child(const path_type &path,
                           ^
  /opt/homebrew/include/boost/property_tree/ptree.hpp:267:20: note: candidate function not viable: 'this' argument has type 'const pinocchio::mjcf::details::ptree' (aka 'const basic_ptree<basic_string<char, char_traits<char>, allocator<char>>, basic_string<char, char_traits<char>, allocator<char>>>'), but method is not marked const
          self_type &get_child(const path_type &path, self_type &default_value);
                     ^
  /opt/homebrew/include/boost/property_tree/ptree.hpp:261:20: note: candidate function not viable: requires single argument 'path', but 2 arguments were provided
          self_type &get_child(const path_type &path);
                     ^
  /opt/homebrew/include/boost/property_tree/ptree.hpp:264:26: note: candidate function not viable: requires single argument 'path', but 2 arguments were provided
          const self_type &get_child(const path_type &path) const;
                           ^
  /tmp/pinocchio-20240527-10090-tqhq72/pinocchio-3.0.0/src/parsers/mjcf/mjcf-graph.cpp:40:19: error: no viable conversion from 'void' to 'pinocchio::mjcf::details::ptree' (aka 'basic_ptree<basic_string<char, char_traits<char>, allocator<char>>, basic_string<char, char_traits<char>, allocator<char>>>')
              ptree attr_parent = parent.get_child(path, ptree());
                    ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

</details>

full build log,https://github.com/Homebrew/homebrew-core/actions/runs/9260490782/job/25474393877
relates to Homebrew/homebrew-core#172967

jcarpent, 2024-06-03:
We have fixed this issue on the devel branch. Could you try it?

chenrui333, 2024-06-11:
@jcarpent can you point me the commits or pr?

jcarpent, 2024-06-11:
See https://github.com/stack-of-tasks/pinocchio/pull/2255

chenrui333, 2024-06-12:
let me check it out, thanks!

jcarpent, 2024-06-25:
I will close this PR. @chenrui333 Feel free to reopen it in case of issue.

closed 2024-06-25 by jcarpent