stack-of-tasks/pinocchio #2277: test-cpp-algorithm-utils-force failure on aarch64-linux with nix

nim65s, 2024-06-06:
Hi,

This is a low priority report, but we're going to disable one test on nix under aarch64-linux, because it fails: https://hydra.nixos.org/build/262360337/nixlog/1
```
78/91 Test #79: test-cpp-algorithm-utils-force ..............***Failed    0.01 sec
Running 1 test case...
/build/source/unittest/algorithm/utils/force.cpp(57): error: in "Test/test_force": check changeReferenceFrame(placement, f_in, LOCAL_WORLD_ALIGNED, WORLD) == f_sol_w has failed

*** 1 failure is detected in the test module "Test"
```

If someone can see an fast and simple way to fix it, don't hesitate, but otherwise I'll work on it, at least to provide a simple way to reproduce it without any need for a full nix setup.

ref. https://github.com/NixOS/nixpkgs/pull/315303

jcarpent, 2024-06-06:
If there is solution like a docker image we might give a try.

jcarpent, 2024-06-06:
It might be just an issue with numerical issues.

jorisv, 2024-06-07:
@nim65s While stabilizing pinocchio I had some similar issues in the conda-forge build farm that have custom CXXFLAGS.
Do you know if the Nix build farm have custom CXXFLAGS also ?

nim65s, 2024-06-12:
@jorisv : Here are the environment variables for this build: [env-vars.txt](https://github.com/user-attachments/files/15805351/env-vars.txt)
Here is the full build command:
```
cd /tmp/nix-build-pinocchio-3.0.0.drv-0/source/build/unittest/algorithm/utils && /nix/store/bqwjxqdz2ywf5zy5lw7wbbkbswwbb26x-gcc-wrapper-13.2.0/bin/g++ -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MODULE=Test -DPINOCCHIO_ENABLE_TEMPLATE_INSTANTIATION -DPINOCCHIO_MODEL_DIR=\"/tmp/nix-build-pinocchio-3.0.0.drv-0/source/models\" -I/tmp/nix-build-pinocchio-3.0.0.drv-0/source/build -I/tmp/nix-build-pinocchio-3.0.0.drv-0/source/build/include -I/tmp/nix-build-pinocchio-3.0.0.drv-0/source/include -I/tmp/nix-build-pinocchio-3.0.0.drv-0/source/unittest/algorithm/utils -isystem /nix/store/zfjwvcycdq7zp972xgnr3hd4wq3q5gyx-eigen-3.4.0/include/eigen3 -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -std=gnu++17 -MD -MT unittest/algorithm/utils/CMakeFiles/test-cpp-algorithm-utils-force.dir/force.cpp.o -MF CMakeFiles/test-cpp-algorithm-utils-force.dir/force.cpp.o.d -o CMakeFiles/test-cpp-algorithm-utils-force.dir/force.cpp.o -c /tmp/nix-build-pinocchio-3.0.0.drv-0/source/unittest/algorithm/utils/force.cpp
```
And here the linking command:
```
/nix/store/bqwjxqdz2ywf5zy5lw7wbbkbswwbb26x-gcc-wrapper-13.2.0/bin/g++ -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG "CMakeFiles/test-cpp-algorithm-utils-force.dir/force.cpp.o" -o test-cpp-algorithm-utils-force  -Wl,-rpath,/tmp/nix-build-pinocchio-3.0.0.drv-0/source/build/src ../../../src/libpinocchio_default.so.3.0.0 /nix/store/ry00r7cmm95xcxkf5qsqhimhdlnqkkfp-boost-1.81.0/lib/libboost_unit_test_framework.so /nix/store/ry00r7cmm95xcxkf5qsqhimhdlnqkkfp-boost-1.81.0/lib/libboost_serialization.so
```

Can you spot any of those bad flags here ?