stack-of-tasks/pinocchio #2289: Unable to compile code with `pinocchio::computeCollisions()` with ROS 2 Humble binary install

scastro-bdai, 2024-06-18:
## Bug description

I've installed `ros-humble-pinocchio` package (it's version 2.6.21) which claims to have HPP-FCL support. However, I don't seem to be able to use collision checking functions.

## Expected behavior

Compiling code that calls `computeCollisions()` should work.

## Reproduction steps

I've successfully written a good deal of code, but when I try to bring in the `computeCollisions()` functions I get a build error:

```
error: ‘computeCollisions’ is not a member of ‘pinocchio’
```

I have included the header: `#include "pinocchio/algorithm/geometry.hpp"`, which appears to be the right one for 2.6.21.

Am I missing something obvious, or is there an issue?

### System

- OS: Ubuntu 22.04
- Pinocchio version: 2.6.21 (from binaries)

jcarpent, 2024-06-19:
@wxmerkt @nim65s Any clue on this issue?

nim65s, 2024-06-19:
I'll be able to look into this next week, if @wxmerkt doesn't have any spare time before that

jcarpent, 2024-06-19:
@scastro-bdai Could you provide reproducible command lines to try quickly in a docker image?

sea-bass, 2024-06-19:
Switching to my personal account, but it's still me!

I've made a simple repo here: https://github.com/sea-bass/pinocchio_ros_cpp_example

As stated in the repo README, you should be able to do:

```shell
docker compose build
docker compose run base
```

Once you're in the container, you can do `colcon build` to build the package, which should work fine.

Now go into the file `pinocchio_ros_example/src/pinocchio_example.cpp` and uncomment the line that calls `computeCollisions()` and you'll find my error:

```
/workspace/src/pinocchio_ros_example/src/pinocchio_example.cpp:16:28: error: ‘computeCollisions’ is not a member of ‘pinocchio’
   16 |   auto result = pinocchio::computeCollisions(model, data, collision_model, collision_data, q);
```

wxmerkt, 2024-06-22:
Checking the binary build log shows that it compiles with `PINOCCHIO_WITH_HPP_FCL` and also runs the collision tests successfully: https://build.ros2.org/job/Hbin_uJ64__pinocchio__ubuntu_jammy_amd64__binary/95/consoleFull

Debug notes:

Running the standard Docker image, installing Pinocchio from binaries shows HPP_FCL is included:
```bash
docker run -it --rm ros:humble /bin/bash
apt update && apt install -y ros-humble-pinocchio
python3 -c "import pinocchio ; print(pinocchio.WITH_HPP_FCL)"
```

When using the repro example (thank you!), it shows that it cannot find hpp-fcl during configuration. This is because it's not included in the exported dependencies, e.g. adding
```cmake
message (WARNING ${pinocchio_LIBRARIES})
```
shows that we are not linking to or exporting the dependency to hpp-fcl.

However, we can see that the `pinocchioTargets.cmake` includes hpp-fcl for the `pinocchio::pinocchio` target.

So with this workaround, it compiles and runs:
```
ament_target_dependencies(pinocchio_example Eigen3 eigen3_cmake_module)
target_link_libraries(pinocchio_example PUBLIC pinocchio::pinocchio)
```

I.e. linking directly using CMake rather than using ament. I am not too familiar with ament so cannot say where exactly the packaging issue comes in.

I fixed your example with this workaround and now it works fine: https://github.com/wxmerkt/pinocchio_ros_cpp_example/commit/d000ca64c1b7f44838ee48357f6de4d6da3935b4

sea-bass, 2024-06-22:
Brilliant, thank you for looking into this!

I also don't know much about the inner workings of ament, but this workaround will for sure let us carry on with our work, without the need to build Pinocchio from source.

nim65s, 2024-06-24:
Thanks @wxmerkt !

sea-bass, 2024-06-27:
By the way... I fleshed out the example in https://github.com/sea-bass/pinocchio_ros_cpp_example, so now it runs and hopefully serves as an example for people using ROS 2 with Pinocchio.

If there is any interest in this going in the regular documentation here, I'm happy to donate this example so it doesn't live on my personal account.

sea-bass, 2024-12-04:
Hi @wxmerkt -- not sure if you prefer a new issue, but I just got sent this (https://github.com/sea-bass/pinocchio_ros_cpp_example/issues/1), and can also reproduce it locally.

So it seems like on Ubuntu 24.04, the same problem still persists even when we do `target_link_libraries`?