stack-of-tasks/pinocchio #2291: Display issues with `hector` drone in pin3

fabinsch, 2024-06-20:
Since I use pinocchio3 from conda, the `hector` drone from example-robot-data is not displayed properly in meshcat. This is not blocking me, I just thought I raise the issue as maybe others will experience display issues as well. I displayed a couple other quadrupeds and manipulator models from erd and they all look fine.

Steps to reproduce:
```
conda create --name erd_debug python=3.11 pinocchio example-robot-data meshcat-python ipython
```
then in ipython
```
import example_robot_data as erd
from pinocchio.visualize import MeshcatVisualizer
robot = erd.load("hector")
vizer =  MeshcatVisualizer(
    robot.model, robot.collision_model, robot.visual_model
)
vizer.initViewer(loadModel=True, open=True)
vizer.display(robot.q0)
```

It works as expected with 
```
conda create --name erd_debug python=3.11 pinocchio==2.7.1 example-robot-data meshcat-python ipython
```
[image]
[image]

jorisv, 2024-06-20:
I think @ManifoldFR have found a similar issue with the UR5 end effector.

ManifoldFR, 2024-06-20:
What is the file extension for the meshes for this robot ?

If it's STL or DAE I have a fix over on our own fork of meshcat-python on the Simple-Robotics org

fabinsch, 2024-06-20:
You are right, the UR5 end-effector has also display issues. However, they look different than what we see on your PR @ManifoldFR .

The drone has [both](https://github.com/Gepetto/example-robot-data/blob/master/robots/hector_description/robots/quadrotor_base.urdf) DAE for visual and STL for collisions. I installed the latest master from [our fork](https://github.com/Simple-Robotics/meshcat-python) of meshcat-python but this does not solve the problem.

[image]

ManifoldFR, 2024-06-20:
Oh, the branch to install is not master, but topic/update-meshcatjs I think.

The current master on the fork is just the master from upstream if I recall correctly.

ManifoldFR, 2024-06-20:
That one should point to the right branch on our fork of meshcat (JS) which has all of the patches I wanted to upstream. Unfortunately nothing has advanced on that front.

fabinsch, 2024-06-20:
mmh, I tried also the branch `topic/update-meshcatjs` with commit `819526d46f9c63efad7a0f3df9a15db0cc2ce0c9` but it looks still the same.

fabinsch, 2024-06-21:
The issue is solved for me by #2294.

closed 2024-06-21 by fabinsch