stack-of-tasks/pinocchio #2313: Error with buildReducedModel

stephane-caron, 2024-07-04:
I can reproduce the issue from the discussion below using this minimal example:

```py
from robot_descriptions.loaders.pinocchio import load_robot_description

robot = load_robot_description("baxter_description")
model = robot.model
reduced_model = pin.buildReducedModel(model, [], robot.q0)
```

It seems related to the Baxter description. I tried replacing `baxter_description` with `upkie_description` and then `buildReducedModel` returns normally.

### Discussed in https://github.com/stack-of-tasks/pinocchio/discussions/2311

<div type='discussions-op-text'>

<sup>Originally posted by **maymohan** July  2, 2024</sup>
Hello,

I recently started using Pinocchio and am interested in building a reduced model for a single arm of a Baxter robot. 

I followed the example in the wiki. But, I get the following error:
`
ValueError: Several frames match the filter - please specify the FrameType
`

Here is a minimum working example of my code:

```
import pinocchio as pin
import numpy as np

# Build the model
pinocchio_model_dir = "./models/" 
mesh_dir = pinocchio_model_dir
urdf_filename = ("./baxter.urdf")
# Load the urdf model
model, collision_model, visual_model = pin.buildModelsFromUrdf(urdf_filename, mesh_dir)

# Build the reduced model
relevantJoints = ['right_s0',
 'right_s1',
 'right_e0',
 'right_e1',
 'right_w0',
 'right_w1',
 'right_w2']

jointsToLockIDs = []
initialJointConfig = np.array([0, #head
				 0, 0,0, 0.1,0,0,0,  # Left arm				 
				 0, 0,0, 0.1,0,0,0  ])# Right arm

for name in model.names:			
	if not name in relevantJoints:	
		if model.existJointName(name):
			jointsToLockIDs.append(model.getFrameId(name))
		else:
			print('Warning: joint ' + str(name) + ' does not belong to the model!')
			
model_reduced = pin.buildReducedModel(model, visual_model, jointsToLockIDs, initialJointConfig)
```
Here is [Baxter's URDF](https://github.com/RethinkRobotics/baxter_common/blob/master/baxter_description/urdf/baxter.urdf).

I looked around quite a bit and could not find anyway to resolve the issue. I would appreciate any help in resolving the issue. 

Please let me know if I can provide any other information. Thank you for your help! 

 
</div>

Zionshang, 2024-07-14:
I haven't looked at your URDF file in detail, but it's probably because some names of `joint` in the URDF are same as some names of `link`. You need to check the urdf in detail.

maymohan, 2024-07-23:
@Zionshang You were right! 

In case anyone else has the same issue, I changed the joint names of the following joints by adding the suffix "_fixed": head_camera, right_hand, right_hand_camera, right_hand_camera_axis, right_hand_range, right_hand_accelerometer, right_gripper_base, left_hand, left_hand_camera, left_hand_camera_axis, left_hand_range, left_hand_accelerometer, left_gripper_base.

My modified URDF file can be found on [my fork](https://github.com/maymohan/baxter_common/blob/master/baxter_description/urdf/baxter_modified.urdf) of the Rethink SDK.

lucasjinreal, 2025-11-26:
Same error here.

line 249, in get_transform_frame_to_world
    frame_id = self.model.getFrameId(frame)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Several frames match the filter - please specify the FrameType

However, maybe pink would add frame_type support as params when this situation occurs?

INo-xious, 2026-09-30:
Hi, draft #2959 has a fix for this. The ambiguity still reproduces at `4accf2f9` when joint and body frames share a name, including with an empty list of joints to lock. The patch preserves frame types during lookup and tests parent relationships and placements.

Could this issue be marked `ready`, and is it available for me to take? The draft's CI workflows also need maintainer approval; `build_collision` would add coverage for the geometry-reduction overloads.

AI assistance was used for this contribution, as disclosed in the PR.