stack-of-tasks/pinocchio #2317: Issue with importing pinocchio.casadi

waynekyrie, 2024-07-08:
## Bug description

I installed Pinocchio following the instructions here [https://stack-of-tasks.github.io/pinocchio/download.html](url) using robotpkg on a Ubuntu 20.04 machine with python 3.8. But when I do `import pinocchio.casadi`, it gives the error msg saying no module named pinocchio.casadi.

Is there anything else that needs to be done? Thank you.

fabinsch, 2024-07-09:
Hi @waynekyrie,
yo simplify, you can create a new conda environment and install Pinocchio using:
```
conda create --name myenv
conda activate myenv
conda install pinocchio -c conda-forge
```
Afterward, you can import `pinocchio.casadi`. Alternatively, building Pinocchio from source is also an option.

nim65s, 2024-07-09:
casadi support is not yet enabled on robotpkg. It will be in the following weeks.

waynekyrie, 2024-07-10:
Thanks for the prompt reply! Is it the same case that casadi is not supported when using homebrew to install on Mac OS X?

fabinsch, 2024-07-10:
> casadi is not supported when using homebrew to install on Mac OS X

correct.

waynekyrie, 2024-07-11:
Thanks so much. It is resolved by using conda.

A follow-up question. After I import pinocchio.casadi, I want to use the rnea() function to calculate the inverse dynamics. However, my pos, vel, and acc are in the type of Casadi.MX, whereas rnea() takes Casadi.SX (if I understand correctly). In this case, is it still possible to use rnea()?

jcarpent, 2024-07-12:
You should use SX and not MX with Pinocchio, as it is much more efficient.

closed 2024-07-12 by jcarpent