stack-of-tasks/pinocchio #2319: Unexpected for Rotation of operationnel frame placements  (oMf)

hubernikus, 2024-07-10:
I get unexpected values in the rotation matrix when computing the kinematics of the robot. 
The values are surprising as the 3x3-rotational matrix elements, have values which are well above 1. 
It seems the rotational matrix takes on the values of the joint angle, rather than the cos/sin of it. 

Am I not using the method correctly? Or is this faulty behavior?

I run the URDF model using the forward kinematics, and subsequential updating of the frame placements, as follows:
```cpp
std::vector<std::optional<double>> joint_configuration = {0, 0, 0, 0, 0, 0, 10};
pinocchio::Data data(this->model);
pinocchio::forwardKinematics(this->model, data, joint_configuration);
pinocchio::updateFramePlacements(this->model, data);

std::cout << "Translation\n " << data.oMf[frame_id].translation().transpose() << std::endl;
std::cout << "Orientation\n " << data.oMf[frame_id].rotation() << std::endl;
std::cout << "Matrix\n " << data.oMf[frame_id].toHomogeneousMatrix() << std::endl;
```

I get values of:
```bash
Translation
  1.14673  1.26659 0.898112
Orientation
   0.00569599 -3.14417e-08 -7.35662e-07
    -9.99999  5.51997e-05   0.00129154
   0.0010171 -5.61436e-09 -1.31363e-07
Matrix
   0.00569599 -3.14417e-08 -7.35662e-07      1.14673
    -9.99999  5.51997e-05   0.00129154      1.26659
   0.0010171 -5.61436e-09 -1.31363e-07     0.898112
           0            0            0            1
```

## System Specifications
Ubuntu 22.04.4 LTS (Docker)
C++ 20 / g++ 11.4.0
Cmake 3.22.1
Boost 1.83.0
Pinocchio: 3.0.0
Eigen: 3.4.0-2ubuntu2

hubernikus, 2024-07-11:
It seems the issue is resolved if I change the `continuous` joints to `revolute` joints. 
With the `continuous` the kinematics do not work.

jorisv, 2024-07-11:
Hello @hubernikus,

continuous have two parameter (cos(theta), sin(theta)) where revolute joint only have one.
Here an explanation: https://github.com/stack-of-tasks/pinocchio/issues/794#issuecomment-490050573

hubernikus, 2024-07-11:
Hello @jorisv  
Thanks a lot for the info. That explains the behavior!

closed 2024-07-11 by hubernikus