stack-of-tasks/pinocchio #2320: Install previous release through robotpkg

Cfather, 2024-07-10:
Hi,

I mainly use Pinocchio in C++ and I usually try to avoid conda in C++ development. 
I usually install Pinocchio through robotpkg as advised in the [documentation](https://stack-of-tasks.github.io/pinocchio/download.html).
I have been putting all my development in the docker until recently I found that Pinocchio was upgraded to 3.0.0/3.1.0.
I was trying to rebuild my docker environment and found that robotpkg automatically installed the latest version of Pinocchio and my cmake compilation script did not work anymore.

I wonder if there's a way for me to install previous release, like 2.7.1, through robotpkg?
I assume that 3* represents the version number so I have tried something like `sudo apt install -qqy robotpkg-py2.7.1-pinocchio` but it didn't work.

I would appreciate it if anyone could help me with this!

nim65s, 2024-07-11:
Hi,

The number after `-py` is the python version, not pinocchio version. so on ubuntu 24.04, there is python 3.12, so `robotpkg-py312-pinocchio` will install pinocchio 3. if you want pinocchio 2, this can work with eg.
`apt install robotpkg-py312-pinocchio=2.7.0 robotpkg-pinocchio=2.7.0 robotpkg-py312-eigenpy=3.5.0`

That being said, we strongly advice not to rely for too long on older versions: we won't provide support for those, and if anything breaks with this pinocchio=2.7.0 we certainly won't fix it.

We encourage you to update your CMake for the latest version. For that, you can follow the upgrade guide: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc__porting.html, and if you have any trouble doing it we can help with that.

Cfather, 2024-07-11:
Thanks for your response!

closed 2024-07-11 by Cfather