stack-of-tasks/pinocchio #2321: Minimal cmake example to compile a C++ program from outside for Pinocchio 3 / undefined reference to parseRootTree

Cfather, 2024-07-10:
## Bug description

Main error when I use cmake to compile a program with pinocchio. The following function can not be found:
```
undefined reference to `pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)
```
The exact same CMakeLists used to work for me a few months ago, until recently I rebuilt my container and found robotpkg automatically installed Pinocchio 3 and the compilation did not work anymore.

I am currently working on a deadline. 
It would be greatly appreciated if anyone could provide a minimal CMakeLists.txt that solves this problem.

## Expected behavior

The test script should be successfully compiled.

## Reproduction steps

You can copy and paste the test script (test.cpp) and the CMakeLists.txt somewhere, then run the following commands:
1. `mkdir build`
2. `cd build`
3. `cmake ..` (this step is good to me)
4. `make` (this step raises error)

### Docker image

If this is an installation issue, a Docker image to reproduce it:
I won't provide my full dockerfile here but I was installing Pinocchio as advised in its [documentation page](https://stack-of-tasks.github.io/pinocchio/download.html) through robotpkg.

```
mkdir -p /etc/apt/keyrings
curl http://robotpkg.openrobots.org/packages/debian/robotpkg.asc \
    | tee /etc/apt/keyrings/robotpkg.asc
echo "deb [arch=amd64 signed-by=/etc/apt/keyrings/robotpkg.asc] http://robotpkg.openrobots.org/packages/debian/pub $(lsb_release -cs) robotpkg" \
    | tee /etc/apt/sources.list.d/robotpkg.list
apt-get update
apt install -qqy robotpkg-py3*-pinocchio

export PATH=/opt/openrobots/bin:$PATH
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/opt/openrobots/lib/python3.10/site-packages:$PYTHONPATH
export CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH
```

### Code

The script and the CMakeLists.txt are attached here to reproduce the issue:

The test script (named as "test.cpp"):
```
#include "pinocchio/parsers/urdf.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"

int main() {
	pinocchio::Model model;
    pinocchio::urdf::buildModel("some_invalid_filename.urdf", model);
	return 0;
}
```

The CMakeLists.txt:
```
# Minimum required CMake version
cmake_minimum_required(VERSION 3.18)

# Project name
project(URDFPARSER
    DESCRIPTION "Test pinocchio urdf parser")

# Set C++ standard
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED True)

# Find Eigen library
find_package(Eigen3 3.3.7 REQUIRED NO_MODULE)
include_directories(${EIGEN3_INCLUDE_DIR})

# Find Boost library
find_package(Boost REQUIRED COMPONENTS
    system
    filesystem
    serialization
)
include_directories(${Boost_INCLUDE_DIRS})
link_directories(${Boost_LIBRARY_DIRS})

# Find Pinocchio library
find_package(PkgConfig REQUIRED)
pkg_check_modules(PKG_PIN_CONFIG REQUIRED 
    pinocchio
    urdfdom
    console_bridge)
include_directories(${PKG_PIN_CONFIG_INCLUDE_DIRS})
link_directories(${PKG_PIN_CONFIG_LIBRARY_DIRS})

set(PINOCCHIO_FLAGS
  ${pinocchio_CFLAGS_OTHER}
  -Wno-ignored-attributes
  -Wno-invalid-partial-specialization    # to silence warning with unsupported Eigen Tensor
  -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR
  -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR
)

add_executable(test 
    test.cpp)

target_link_libraries(test PRIVATE
    ${BOOST_LIBRARIES} 
    ${PKG_PIN_CONFIG_LIBRARIES})

target_compile_options(test PUBLIC
    ${PINOCCHIO_FLAGS})
```

## Additional context

Full error message on my side:
```
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>& pinocchio::urdf::buildModel<double, 0, pinocchio::JointCollectionDefaultTpl>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, bool)':
test.cpp:(.text._ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b[_ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b]+0x5b): undefined reference to `pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test.dir/build.make:97: test] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/test.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

### System

- OS: Ubuntu 22.04
- Pinocchio version: 3.1.0

Cfather, 2024-07-10:
BTW, I have checked [https://github.com/stack-of-tasks/pinocchio/issues/1574](https://github.com/stack-of-tasks/pinocchio/issues/1574) for multiple times and my CMakeLists.txt was following [this response](https://github.com/stack-of-tasks/pinocchio/issues/1574#issuecomment-1766229126). That's why it worked at the first time.

jorisv, 2024-07-11:
Hello @Cfather,

Since Pinocchio 3, the library had been split in many sublibraries ([see CMake Targets](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc__porting.html)).

If you used find_package(pinocchio) the pinocchio target link on all the sublibraries for backward compatibility.

I have run `pkg-config --libs pinocchio` and I don't see pinocchio_parsers inside it. I will check why the .pc file doesn't contains it.

In the meantime, you can use find_package or try to add `pinocchio_parsers` as target_link_libraries argument.

nim65s, 2024-07-11:
Thanks for this detailed report. 
FWIW, we also have an minimal CMake example here: https://github.com/stack-of-tasks/pinocchio-minimal
But I'm not sure it is up to date with pinocchio 3.

cmastalli, 2024-07-11:
> Hello @Cfather,
>
> Since Pinocchio 3, the library had been split in many sublibraries ([see CMake [...]

Hi @jorisv 

We are having a failure in our Crocoddyl's CONDA CI that might be related. This compilation error recently happened in this PR: https://github.com/loco-3d/crocoddyl/pull/1274, where you can find more details at https://github.com/loco-3d/crocoddyl/actions/runs/9847244715/job/27189046081?pr=1274 or in the copied message below.

Could you advise me on how to fix it? (Thanks)

```bash
-- generate_stubs: adding dependency on crocoddyl_pywrap. Stubs will be generated after it is built.
-- Configuring done (14.7s)
CMake Error at /Users/runner/miniconda3/envs/crocoddyl/lib/cmake/pinocchio/pinocchioTargets.cmake:79 (set_target_properties):
  The link interface of target "pinocchio::pinocchio_extra" contains:

    Qhull::qhullcpp

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /Users/runner/miniconda3/envs/crocoddyl/lib/cmake/pinocchio/pinocchioConfig.cmake:182 (include)
  cmake/package-config.cmake:113 (find_package)
  CMakeLists.txt:106 (add_project_dependency)
```

nim65s, 2024-07-11:
@cmastalli this is waiting for https://github.com/conda-forge/qhull-feedstock/pull/19

nim65s, 2024-07-11:
> But I'm not sure it is up to date with pinocchio 3, I will double check that.

Yes, it is still working, from pinocchio 2.6.11 on ROS to 3.0.0 on nix and robotpkg.

Cfather, 2024-07-11:
> > But I'm not sure it is up to date with pinocchio 3, I will double check that.
> 
> Yes, it is still working, from pinocchio 2.6.11 on ROS to pinocchio 3.0.0 on nix and robotpkg.

Thanks for the reply. I have solved my problem by referring to this example!

Instead of using pkgconfig, simply use 
```
find_package(pinocchio REQUIRED)
include_directories(${PINOCCHIO_INCLUDE_DIRS})
link_directories(${PINOCCHIO_LIBRARY_DIRS})
```
to find the whole package and include them in the path.

And in `target_link_libraries`, instead of using `${pinocchio_LIBRARIES}`, simply use `pinocchio::pinocchio`.

But I'm not sure why pkgconfig stops working though.

nim65s, 2024-07-11:
Glad it works, thanks for the update !

Ideally, you should not even need `include_directories(${PINOCCHIO_INCLUDE_DIRS})` and `link_directories(${PINOCCHIO_LIBRARY_DIRS})`: those informations are encoded in a modern and relocatable way in `pinocchio::pinocchio`.

pkgconfig issue should be fixed after #2322.

Cfather, 2024-07-11:
Thanks!
I'm good to close this issue now.

closed 2024-07-11 by nim65s