stack-of-tasks/pinocchio #2324: Installing Pinocchio 2.7.0 on arm64 architecture

Tabi43, 2024-07-11:
Hi, 
I’m working with an nvidia Jetson nano (arm64 architecture) inside which I have a Docker container based on the official image of ros:melodic. I would need the version of Pinocchio 2.7.0 but the apt repository of robotpkg has amd binaries only, ros-melodic-pinocchio is only available in version 2.6.16, I tried to compile it from source using robotpkg but I had several of problems compiling the hpp-fcl library. Do you have any suggestions to correctly inlay this library ?

jcarpent, 2024-07-11:
Could you use conda instead?

Tabi43, 2024-07-11:
I would use the library in c++, if I’m not wrong with conda I would only have the python API of Pinocchio, right?

ManifoldFR, 2024-07-11:
> I would use the library in c++, if I’m not wrong with conda I would only have the python API of Pinocchio, right? Sorry, I forgot to mention it in the issue.

Not at all mate, you're in fact getting the whole package with the CMake targets and C++ headers installed in your Conda prefix

Tabi43, 2024-07-11:
Great, then I’ll try to use conda. Thank you very much, i will update you.

jcarpent, 2024-07-12:
Will close this issue. @Tabi43 Feel free to reopen it in case of need.

closed 2024-07-12 by jcarpent