stack-of-tasks/pinocchio #2325: Issues with isaacgym

Norweig1an, 2024-07-11:
Hi,
We are using the project in the issacgym environment, but there is an error:
When we read the model from urdf and run `print(model.names)`
after `import isaacgym`
we encountered an error:
`TypeError: No Python class registered for C++ class std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >`

However, if we do the above steps before `import isaacgym`, the error does not occur.

Would you please give us some ideas about how to deal with it? Our python version is 3.8.19 and our isaacgym version is 1.0rc4.

ManifoldFR, 2024-07-11:
Hello, thanks for your interest in Pinocchio!

Could you provide a minimum example?

I'm not sure how Isaac Gym exposes things from C++ to Python, I'd wager it's using pybind11 but I can't be sure. Pinocchio uses the older Boost.Python, which has a conversion registry system, and in the bindings to Python we define a bunch of wrapper classes for common C++ classes (here, `std::vector<std::string>`). The import order (which package you load first) can change which wrapper classes or converters for many classes are defined first and at the "top" of the "conversion registry".
It's important to check your import order: either you need to import Pinocchio first, or IsaacGym first, to get consistent behaviour.

jorisv, 2024-07-12:
@ManifoldFR I have downloaded isaacgym package and it package libboost_python. The gym_38.so link against boost_python, so you're probably right. Something weird it that a ton of distributed library also link against libboost_python but are not python modules…

We already encounter similar issue with ompl : https://github.com/stack-of-tasks/eigenpy/issues/392
But it's look like isaacgym is more aggressive with the registry…

ManifoldFR, 2024-07-12:
Huh, weird that Isaac gym is using Boost.Python... seems like we're not the last ones

jcarpent, 2024-07-21:
@Norweig1an Unfortunately, it seeems that the way isaacgym packages boost-python is faulty.
We cannot do much as we are not the maintainer of this package.

As you have found a solution (by proper import) to circumstantiate this issue, we will close it.

Thanks for raising it and providing a solution to the Pinocchio community.

closed 2024-07-21 by jcarpent