stack-of-tasks/pinocchio #2328: Error in pinocchio::appendModel()

Zionshang, 2024-07-14:
## Bug description
I'm trying to combine  two models into a new one by `appendModel()` method. But I get the following error

```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  The two models have conflicting frame names.
```
I'm pretty sure that the two models don't have conflicting frame names. **Because when I manually combined URDF files of these two models together, there was no problem**

## Code

```C++
int main()
{
    std::string manipulatorUrdfPath = projectPath + "/urdf/manipulator.urdf";
    pinocchio::Model manipulator;
    pinocchio::urdf::buildModel(manipulatorUrdfPath, manipulator, false);
    std::cout << manipulator << std::endl;
    std::string quadrupedUrdfPath = projectPath + "/urdf/quadrupedRobot.urdf";
    pinocchio::Model quadruped;
    pinocchio::urdf::buildModel(quadrupedUrdfPath, quadruped, false);
    std::cout << quadruped << std::endl;

    pinocchio::Model quadrupedalManipulator;
    pinocchio::JointIndex idTrunk = quadruped.getFrameId("trunk", pinocchio::BODY);
    std::cout << idTrunk << std::endl;
    pinocchio::appendModel(quadruped, manipulator, idTrunk, pinocchio::SE3::Identity(), quadrupedalManipulator);
}
```

The output is

```
Nb joints = 2 (nq=1,nv=1)
  Joint 0 universe: parent=0
  Joint 1 joint1: parent=0

Nb joints = 2 (nq=7,nv=6)
  Joint 0 universe: parent=0
  Joint 1 floating_base: parent=0

4
terminate called after throwing an instance of 'std::invalid_argument'
  what():  The two models have conflicting frame names.
```

The simplified URDF file `manipulator.urdf` is:

```xaml
<robot name="z1_description">
  <link name="link00">
  </link>
  <joint name="joint1" type="revolute">
  </joint>
  <link name="link01">
  </link>
</robot>
```

The simplified URDF file `quadrupedRobot.urdf` is:

```xaml
<robot name="aliengo">
  <link name="base">  
  </link>
  <joint name="floating_base" type="floating"> 
  </joint>
  <link name="trunk"> 
  </link>
  <joint name="imu_joint" type="fixed"> 
  </joint>
  <link name="imu_link">
  </link>
</robot>
```
**Judging by the URDF, they don't have the same frame name**
Is it because they all have `universe joint 0`, so there is the error?

I've been haunted for a long time and I would be so grateful if you could help me!

Zionshang, 2024-07-14:
By the way, I can only print out the names of `FrameType::JOINT` now. How do I print names of all frames, including `FrameType::BODY`, `FrameType::FIXED_JOINT`?

Zionshang, 2024-07-14:
I printed out all the frame names for these two models.
```
--------------name of manipulator frame-- -- -- -- -- -- --
Frame 0: universe
Frame 1: root_joint
Frame 2: link00
Frame 3: joint1
Frame 4: link01

--------------name of quadruped frame-- -- -- -- -- -- --
Frame 0: universe
Frame 1: root_joint
Frame 2: base
Frame 3: floating_base
Frame 4: trunk
Frame 5: imu_joint
Frame 6: imu_link
```
It seems that `root_joint` caused the error.
The question is, why did root_joint appear? I don't define them in the URDF

Zionshang, 2024-07-14:
Is there a `root_joint` every time when loading a model from URDF? If so, I think it will never be possible to combine two URDFs model together via `appendModel`. :(

jcarpent, 2024-08-26:
@MegMll Could you handle this issue?

MegMll, 2024-08-26:
I will look into it this week

jorisv, 2024-09-26:
Hello @Zionshang,

#2402 had been merged. I close this issue.

closed 2024-09-26 by jorisv