stack-of-tasks/pinocchio #2329: newer version meshcat visualization issue for mesh

MiaoDragon, 2024-07-15:
Hello. Thanks for the software effort of the Pinocchio. I've been experiencing a visualization issue in meshcat viewer for meshes. It seems for the new version 3.1 the mesh scale is not correctly displayed when I have a mixed mesh files of STL and DAE. One visualization is shown below for my motoman robot with robotiq gripper, and the gripper has a very large scale. This is the visual geometries.
[image]
The collision geometries are also problematic.
[image]

The scale issue of visual geometrices do not exist for version 2.7, as shown below:
[image]
However, the collision geometries still have the same issue.

The visualized model can be extracted from
https://github.com/DanManN/motoman

After the model package is installed in a catkin workspace, and the workspace is made, use the following code to visualize the results:
(the following code is saved as a file motoman_config.py)
`
import sys
import os
import rospy
from rospkg import RosPack

class MotomanSDA10F:
    def __init__(self):
        self.ignore_collision_ee_links = [
            "motoman_right_ee",
            "left_outer_knuckle",
            "left_outer_finger",
            "left_inner_finger",
            "left_inner_finger_pad",
            "left_inner_knuckle",
            "right_outer_knuckle",
            "right_outer_finger",
            "right_inner_finger",
            "right_inner_finger_pad",
            "right_inner_knuckle",
            "robotiq_arg2f_extra_link",
            "robotiq_arg2f_base_link",
            "arm_right_link_7_t",
            "arm_right_link_6_b",
        ]
        rp = RosPack()
        self.robot_urdf = rp.get_path(
            'motoman_sda10f_moveit_config'
        ) + '/config/gazebo_motoman_sda10f.urdf'
        package_dirs = []
        for package in ['motoman_sda10f_support', 'robotiq_2f_85_gripper_visualization']:
            path = os.path.join(rp.get_path(package), '..')
            path = os.path.abspath(path)
            package_dirs.append(path)
            # package_dirs.append(rp.get_path(package))
        self.package_dirs = package_dirs
`

`
import motoman_config
import pinocchio as pin
import hppfcl
from pinocchio.visualize import MeshcatVisualizer
import time
import numpy as np

motoman = motoman_config.MotomanSDA10F()
model, collision_model, visual_model = pin.buildModelsFromUrdf(motoman.robot_urdf, package_dirs=motoman.package_dirs)
viz = MeshcatVisualizer(model, collision_model, visual_model)
viz.initViewer(open=True)

# Load the robot in the viewer.
viz.loadViewerModel()

# Display a robot configuration.
q0 = pin.neutral(model)
print('q0: ')
print(q0)
time.sleep(1.0)
viz.display(q0)
viz.displayVisuals(True)
`
The system information is the following:

- OS: Ubuntu 20.04
- Pinocchio version: 3.1.0 and 2.7.0
- meshcat: 0.3.2

Thanks and any help would be appreciated!

jcarpent, 2024-07-16:
Will have a look soon.

jcarpent, 2024-07-17:
Solved via #2331.

closed 2024-07-17 by jcarpent